StevenHickson/PiAUISuite #43: InstallAUISuite fails

black-snow, 2015-07-24:
InstallAUISuite.sh =>

```
Would you like voicecommand to try to set itself up? y/n
y
./InstallAUISuite.sh: line 186:  1244 Illegal instruction     voicecommand -s
```

Google didn't help so I thought I'd ask here. Tried to install it on my old raspi B+.

alecgibson, 2015-07-27:
I confirm that I get the same issue running a B+ running Minibian.

I'm wondering if it's to do with the updates to make PiAUISuite compatible with the Pi v2? The Pi 2 uses ARMv7, and apparently the Pi 1 (ARMv6) incorrectly reports its architecture version as v7?

See a similar issue [here](https://github.com/nodejs/io.js/issues/283#issuecomment-69515154).

black-snow, 2015-07-31:
@alecgibson with the changes you suggested I got further. Sadly I now have another issue. I get to this:

    First I'm going to say something and see if you hear it
    Did you hear anything? (y/n)

But nope, no output. `$ speaker-test -c 2` runs just fine. Any guesses?

black-snow, 2015-08-02:
```
$ voicecommand
Opening config file...
ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
arecord: main:722: audio open error: No such file or directory
Warning: Couldn't read data from file "/dev/shm/out.flac", this makes an empty
Warning: POST.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    14    0    14    0     0     38      0 --:--:-- --:--:-- --:--:--    38
rm: cannot remove '/dev/shm/out.flac': No such file or directory
No translation

```

black-snow, 2015-08-02:
I'll make this a new issue

bruno1308, 2016-02-06:
Anything on this? I can't hear anything when I run voicecommand -s (working on Pi 2)

velkrosmaak, 2016-02-08:
I'm in the same boat as bruno1308 - Pi2, wheezy image, can't hear anything during the voicecommand setup, but audio plays fine with aplay and speaker-test.

agustinscaz, 2016-02-14:
Same problem here, i can't hear anything :-1:

rjturek, 2016-02-20:
Same problem here.  speaker-test and aplay work fine.  Hear nothing during setup. Pi2 wheezy

black-snow, 2016-02-21:
reopening this

perskes, 2016-09-01:
Anything new on this?

Reidmere, 2016-10-15:
So, the reason (for those following) for not hearing anything, is because Google blocked connecting to it for TTS. Someone, however, cleverly created a wrapper that allows audio to play, though it requires installing Pico2Wave.

I have gotten it working, but it kept looping "what you say baby", until I changed what alecgibson suggested. After that, I'm getting the out.flac errors (which I think may actually be a step back).

To get the audio working, I'm going to reuse the standard install and see if I can't debug what's causing her to mistake the clip for "useless conversation", so to speak, later...

Reidmere, 2016-10-16:
I've done some digging, and the bug is on line 35 of VoiceCommand/speech-recog.sh

I'm looking into seeing if I can make a reasonable replacement using Pico2Wave.
I don't really know the infrastructure on how it works before that point, but I'm hoping I can just reiterate the speech string to play through tts -> Pico2Wave through a wrapper someone very generously posted elsewhere.

passi1995, 2016-10-29:
Hey, I#ve got the same problem and I can't fix it.
Mic and Speakers are working, I changed the Google API to VoiceRSS.

Can anyone help me please? :/

[image]

BHEADRICK, 2017-01-09:
this is my tts script:
`#!/bin/bash

string=$@
lang="en-us"
echo "" > "/dev/shm/speak.mp3"

len=${#string}

wget -q -U Mozilla -O "/dev/shm/tmp.mp3" "http://api.voicerss.org/?key=<API key>&hl=${lang}&src=$string"
cat "/dev/shm/tmp.mp3" >> "/dev/shm/speak.mp3"
#now we finally say the whole thing
cat "/dev/shm/speak.mp3" | mpg123 - 1>>/dev/shm/voice.log 2>>/dev/shm/voice.log
`

projectclass, 2018-02-23:
Warning: Couldn't read data from file "/dev/shm/noise.flac", this makes an 
Warning: empty POST.
how to solve it?

borhanshuvo, 2019-12-22:
Warning: Couldn't read data from file "/dev/shm/noise.flac", this makes an
Warning: empty POST.