StevenHickson/PiAUISuite #45: Not output from voicecommand while aplay and others run just fine.

black-snow, 2015-08-02:
Hi,

I don't get voicecommand to talk to me. aplay and mpg123 work just fine, arecord, too. !hardware==plughw:0,0

Sometimes I get output like this:
```
Opening config file...
running in continuous mode
verifying command as well
keyword duration is 2 and duration is 3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 36597    0    14  100 36583      9  25085  0:00:01  0:00:01 --:--:-- 25091
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38689    0    14  100 38675      7  19478  0:00:01  0:00:01 --:--:-- 19483
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 40122    0   234  100 39888     81  13942  0:00:02  0:00:02 --:--:-- 13941
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38668    0   185  100 38483     92  19318  0:00:01  0:00:01 --:--:-- 19318
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38144    0    14  100 38130      9  25881  0:00:01  0:00:01 --:--:-- 25903
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 36882    0    14  100 36868      9  25318  0:00:01  0:00:01 --:--:-- 25338
```

But nothing else happens. I don't hear anything during the setup.

Any guesses?

Jeins98, 2015-08-05:
Same issues her. Please help

black-snow, 2015-08-08:
@StevenHickson any suggestions?

JBmiog, 2015-08-09:
Same here, the .mp3 returned from google's text to speech service are 0 bytes in size, I've checked this out and i'm getting a 503 http error when trying to use their TTS service.

StevenHickson, 2015-08-19:
This should be fixed if you do a git pull. Google started using a captcha for that url so I had to modify it slightly. 
Can anyone run UpdateAUISuite.sh or git pull and InstallAUISuite.sh voicecommand and confirm that it is fixed?

Jeins98, 2015-08-20:
Thank you very much. I will test it when I arrive at home

black-snow, 2015-08-21:
Sadly the update didn't fix anything for me. Still no sound while other applications output just fine.

Does the output on the console change when voicecommand has recognized the keyword? I'm still just getting the above console output.

lemme, 2015-08-25:
me too - the update didn't fix the/my problem?!

StevenHickson, 2015-08-26:
Did you run git pull?
Did you run UpdateAUISuite.sh?

It is working fine on multiple computers for me.Can you run just a test with tts:
tts "Testing a phrase 1 2 3"

joepdooper, 2015-08-28:
The UpdateAUISuite.sh worked for me, thanks!

sudo /home/pi/PiAUISuite/Install/UpdateAUISuite.sh

black-snow, 2015-08-29:
Alright, now it's talking to me during install and tts `test` runs fine. But it doesn't seem to recognize the keyword.

Jeins98, 2015-09-02:
Thank you! Everything works fine.

lemme, 2015-09-03:
I have got the same problem as black-snow...
... I run git pull
... I run UpdateAUISuite.sh
... I tested it with tts "Testing a phrase 1 2 3" - sound works fine
... I tested it with tts -l de "Hallo ich bin ein Schweinchen" - sound works fine too
... if I start voicecommand I get:
Opening config file...
Not querying for answers
running in continuous mode
verifying command as well
keyword duration is 3 and duration is 2
... if I say "pi" - nothing happen

Is there a solution?
greetz

joepdooper, 2015-09-03:
Do 'voicecommand -e' and check your config file.

https://github.com/StevenHickson/PiAUISuite/issues/42

black-snow, 2015-09-05:
@lemme guess we should make a new issue for that

closed 2015-09-05 by black-snow

lemme, 2015-09-06:
@joepdooper 
thank you for the hint (#42) but there where no empty lines (I checked ".commands.conf" and "command.conf").

@black-snow 
do you have the same problems? did you start a new issue?

black-snow, 2015-09-10:
@lemme no, I have not start a new one yet - feel free to do so

ccurran689, 2015-12-20:
Has this issue been resolved? I've been trying to get this working for a week with no luck. I've ran all the updates and re-installed it countless times without being able to get any sound during install or when I run the actual program @black-snow @StevenHickson

joepdooper, 2015-12-21:
Did you check your config file for empty lines through 'voicecommand-e'?
https://github.com/StevenHickson/PiAUISuite/issues/42

ccurran689, 2015-12-21:
I did yes

ccurran689, 2016-01-04:
@joepdooper There seems to be many with the same issue has no one resolved it? I've even tried an older version pi

black-snow, 2016-01-04:
Gotta say that I haven't been trying for a while.

ccurran689, 2016-01-04:
I've tried pretty much everything I can try so just keep and eye out for me if you can

joepdooper, 2016-01-05:
Did it work before the update? Because that is the issue we are commenting on here…

joepdooper, 2016-01-05:
Does $arecord/$aplay work for you?

ccurran689, 2016-01-05:
Yep did indeed

joepdooper, 2016-01-05:
Is your hardware set correct in your config?

ccurran689, 2016-01-05:
Yeah I've used modprobe and tried changing the !hardwareplughw: to a few different settings just to check

scooper86, 2016-01-16:
I too seem to have this issue, although have not yet played around with !hardwareplughw:

voicecommand setup does not recognise anything I say and comes back with symbols.

I am currently running Jessie and have also configured bluetooth a2dp so don't know if I have interfered with anything related to pulseaudio/asla.

wildbi111, 2016-05-24:
I am having a similar issue in that voicecommand and tts do not work, yet aplay, arecord and mpg123 all work.  Any suggestions?