StevenHickson/PiAUISuite #47: No Audio Output

TewGer, 2015-10-22:
Hello,

i have a Problem with the Audio Output of PiAUISuite, because there is no Audio Output. My Speakers are working good, i can Play Audio, but with PiAUISuite there is nothing. For Example by running the Setup, i couldn´t hear anything. 
By Starting PiAUISuite everything was ok, but after some minutes the Audio Output was dead. 

Do you have any Idea about it?

zeneslev, 2015-11-01:
I'm having a similar problem. I was able to run the setup fine, but when I try to use voicecommand -c I don't get any audio output. After checking out the tts scrip that runs to generate the .mp3 file for playback, I believe the issue is with the Google Translate API now only servicing a few requests per day.

RenierD, 2015-11-04:
Hi guys. This is such a useful little suite. Thanks Steve! I've been fiddling with it a lot. During the last month or so I've also been getting this problem. I also believe that it is a limitation from Google. Some days, when I start working on it after a few days I manage to get a about 2 TTS playback then it stops again.

Is there any workararound on this yet? Seems like google want us to register (as developers?) but where and how?

zeneslev, 2015-11-04:
I made a workaround which I documented here:
http://www.modified.systems/archives/68

Hope it helps

kwkevinlin, 2015-11-12:
I'm receiving the same issue here too. I think I heard a response back the first few times I tried it (ex: during setup), but not it's not responding again.

zeneslev, 2015-11-12:
Kevin,

See if this workaround helps:
http://www.modified.systems/archives/68

kwkevinlin, 2015-11-14:
I actually tried that, but I think I'm getting some errors on the PiAUISuite side. Every time I exit out of voicecommand, I have to reboot or else it won't detect my sounds anymore (no more recording activity on the console). Also, when built-in TTS worked for a while, it wouldn't detect "Pi" for some reason. I took off !verify, then it started responding with "Yes Sir?". Do you know what might be the problem?

frozenpandaman, 2015-11-14:
Thank you @zeneslev! You can also just use the "say" command (if you have it installed).

RenierD, 2015-12-24:
Kevin's workaround works well for the TTS side, but it now also seems like the STT side is not working as well. As kwkevinlin said, by deactivating !verify, VoiceCommand responds, but cannot register a command? Is there anyone that has a similar problem?

ccurran689, 2015-12-24:
did you get this resolved @TewGer

TewGer, 2016-01-05:
Yes,

Google only allow a few Connections per day. There is no Audio Output because you don´t get an reply from the Google engine.

promo81, 2016-01-05:
@TewGer i'm trying to put my Trial Google Api Key (Translate API)...you know how?

ccurran689, 2016-01-05:
@tewger so what did you do to fix it ?

TewGer, 2016-01-06:
@promo81 no, i don´t have a key

@ccurran689  i have used espeak instead of the Google api. So you have to edit the tts script. But the Sound and the voice is terrible...

ccurran689, 2016-01-06:
@TewGer Was it just the URL that you adjusted? if not could you post the code change here please

ccurran689, 2016-01-11:
@TewGer ?

TewGer, 2016-01-12:
i will post my code here in a few days, currently i´am not at home. Sorry

zeneslev, 2016-01-15:
Hi All. I am now having problems with STT as well. @TewGer Looking forward to seeing what you did to workaround this.

zeneslev, 2016-01-15:
I managed to fix it using the fix that kmadhugit has posted here: https://github.com/kmadhugit/PiAUISuite/commit/37992bd8a2974b2d19ed6ba9a989a2bf94839dfa

Basically a recent update to the Google SST API requires a different file format. I made these two changes, recompiled (had to change architecture in the makefile though from arm-v7 to arm) and tested. Everything works again!

ccurran689, 2016-01-15:
> kmadhugit@37992bd

Just followed those instructions and I'm still without any output, I can
get output from jasper using espeak however

passiqmukhtar, 2017-05-19:
Hi everyone , I have the same issue above 
any suggestions please