StevenHickson/PiAUISuite #49: No sound output

ccurran689, 2015-12-26:
I am currently trying to get voicecommand working, I've been at it for about a week trying to use jasper also with any luck. I've followed all the step by step guides, all the troubleshooting guides and I have searched the internet trying to find out what the issue is. I have changed the alsa-mixer settings also. I can hear and record things using arecord and aplay commands but I cannot get any sound from the piauisuite voicecommand program it's self, bothing during install when it checks the settings and when I run the actual program. Any help would be greatly appreciated.

ccurran689, 2015-12-26:
I have also changed the !hardwareplughw: to the appropriate setting, when I run voicecommand it will recognize a noise has been made but will not respond or translate.

ccurran689, 2015-12-28:
@woodring63 do you reckon it's got something to do with the pi 2 model b? All the stuff I've seen online are using older versions of the pi

woodring63, 2015-12-28:
Hmm that's possible, I'm also using the pi 2 model b

ccurran689, 2016-01-04:
@woodring63  I got my hands on an older pi, still having same issues. Has there actually been no fix for this?!

woodring63, 2016-01-04:
@ccurran689 Yeah I have no idea, I haven't tried in a couple days but I have yet to figure out what's wrong. I currently have mine accepting commands properly, but still no audio output

ccurran689, 2016-01-04:
Same here keep me updated if you get any further

SweeneyToddTheDemonBarber, 2016-04-03:
Did anyone figure it out?

irsx02, 2016-04-09:
I'm assuming that you already have aplay/arecord working.  (I'm on RaspPi 3, with Raspbian Jessie. but it be the same with RaspPi 2)  
  
`tts  "hello world. can you hear me now" `

If you do not hear your RaspPi speaking, then its because Google changed their TTS application.  Look at article #56 for how to fix it.