StevenHickson/PiAUISuite #50: Stereo vs Mono Microphone

arudd561, 2015-12-26:
It looks like (at least for me) keyword recognition in voicecommand does not work with a mono microphone.  I can change the channels in speech-recog to process mono and that works correctly but keyword verification still tries to capture in stereo.

CoOral, 2016-01-31:
I can't actually get the keyword verification work correctly, even with the kmadhugit fix.
I'm trying to dig out the problem under the code but still no luck.

arudd561, 2016-01-31:
I was able to get it fixed by actually editing the code to force mono recording when listening and recompiling. I have the code set up somewhere on a repository. This didn't use to be a problem so I wonder if there was an update to raspbian that is causing the problem. I can post the changes that I made once I'm back on my computer but I'm traveling right now.

CoOral, 2016-01-31:
I thank you so much for the fast response, as i've tried to fix this in the past two days.
I'll look forward for your post with changes, whenever you'll have time for it.

Respectfully,

Nicola

arudd561, 2016-02-01:
okay Nicola the link to the repo I use that is functioning is https://github.com/arudd561/alfred
I have made a couple of changes in order to get it functioning for me so I would first focus on them one at a time.  The first step was changing the command.cpp file to force mono recording when listening for the keyphrase.  That was changed like you noted on line 36.  Make sure that when you change the file on your Pi you actually delete the voicecommand file and recompile otherwise it wont work.  You cant simply just change the voicecommand.cpp file.  If you have any questions about that let me know.

The next step for me was to change the speech-recog.sh to use mono as well.  You can take a look at my file in the repo but its the same idea by adding -c 1 to force mono.

One last thing I did was switch out google api for a different service because it kept dying on me.  Turns out it was because I threw too many requests and it was trying to captcha me.  I changed it to a free service called voice rss.  You can see that change in the file called tts.  Let me know if you have any questions and if it doesnt work for you.  Happy to help get it working for you

CoOral, 2016-02-01:
First of all i'm gratefull for your time and patience, as you're really helping me out with this issue.
As you might already got, i'm not so capable of fixing it myself 'cause i lack some good skills in programming (sadly, i'm a mechanical eng..sigh).
Anyway, i've followed your changes and deleted the voicecommand file, but i'm having some problems recompiling the voicecommand.cpp:
 
--------------
g++ voicecommand.cpp -o voicecommand

/tmp/ccjWSRch.o: In function `VoiceCommand::~VoiceCommand()':
voicecommand.cpp:(.text+0xc30): undefined reference to `curl_easy_cleanup'
/tmp/ccjWSRch.o: In function `VoiceCommand::Init()':
voicecommand.cpp:(.text+0x1fd4): undefined reference to `curl_easy_init'
voicecommand.cpp:(.text+0x2018): undefined reference to `curl_easy_setopt'
voicecommand.cpp:(.text+0x2030): undefined reference to `curl_easy_setopt'
voicecommand.cpp:(.text+0x2050): undefined reference to `curl_easy_setopt'
voicecommand.cpp:(.text+0x2068): undefined reference to `curl_easy_setopt'
voicecommand.cpp:(.text+0x2080): undefined reference to `curl_easy_setopt'
/tmp/ccjWSRch.o:voicecommand.cpp:(.text+0x2098): more undefined references to `curl_easy_setopt' follow
/tmp/ccjWSRch.o: In function `VoiceCommand::Search(char const*)':
voicecommand.cpp:(.text+0x2eb8): undefined reference to `curl_easy_perform'
voicecommand.cpp:(.text+0x3158): undefined reference to `curl_easy_escape'
voicecommand.cpp:(.text+0x3178): undefined reference to `curl_free'
voicecommand.cpp:(.text+0x31a0): undefined reference to `curl_easy_setopt'
voicecommand.cpp:(.text+0x31c0): undefined reference to `curl_easy_perform'
/tmp/ccjWSRch.o: In function `boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::assign(ch                   ar const*, char const*, unsigned int)':
voicecommand.cpp:(.text._ZN5boost11basic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE6assignEPKcS7_j[_ZN5boost11bas                   ic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE6assignEPKcS7_j]+0x2c): undefined reference to `boost::basic_regex<c                   har, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::do_assign(char const*, char const*, unsigned int)'
/tmp/ccjWSRch.o: In function `bool boost::regex_search<char const*, std::allocator<boost::sub_match<char const*> >, char, boo                   st::regex_traits<char, boost::cpp_regex_traits<char> > >(char const*, char const*, boost::match_results<char const*, std::all                   ocator<boost::sub_match<char const*> > >&, boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> >                    > const&, boost::regex_constants::_match_flags, char const*)':
voicecommand.cpp:(.text._ZN5boost12regex_searchIPKcSaINS_9sub_matchIS2_EEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEEbT                   _SA_RNS_13match_resultsISA_T0_EERKNS_11basic_regexIT1_T2_EENS_15regex_constants12_match_flagsESA_[_ZN5boost12regex_searchIPKc                   SaINS_9sub_matchIS2_EEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEEbT_SA_RNS_13match_resultsISA_T0_EERKNS_11basic_regexI                   T1_T2_EENS_15regex_constants12_match_flagsESA_]+0x84): undefined reference to `boost::re_detail::perl_matcher<char const*, st                   d::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::find()'
/tmp/ccjWSRch.o: In function `boost::re_detail::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boo                   st::regex_traits<char, boost::cpp_regex_traits<char> > >::perl_matcher(char const*, char const*, boost::match_results<char co                   nst*, std::allocator<boost::sub_match<char const*> > >&, boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_                   traits<char> > > const&, boost::regex_constants::_match_flags, char const*)':
voicecommand.cpp:(.text._ZN5boost9re_detail12perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsI                   cEEEEEC2ES3_S3_RNS_13match_resultsIS3_S6_EERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsES3_[_ZN5boost9re_detai                   l12perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEC5ES3_S3_RNS_13match_resultsIS3_S6_E                   ERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsES3_]+0xe0): undefined reference to `boost::re_detail::perl_match                   er<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::                   construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_cons                   tants::_match_flags)'
collect2: error: ld returned 1 exit status
-----------

I know i'm probably making a mistake with the compiling command, but in those hours I couldn't find more informations on the net to fix this.
Also, using the command "locate voicecommand" it throws out that voicecommand is still on the VoiceCommand folder, even after "rm voicecommand" ; i don't know if that is right.
(also, i can locate a voicecommand file under the /usr/bin folder, but i didn't remove it).

Thank you again for the response given, and i hope to help you back in the future if you ever might need it.

Best Regards,

Nicola

arudd561, 2016-02-01:
> undefined reference toboost::basic_regex > >::do_assign(char const*, char
> toboost::re_detail::perl_matcher >, boost::regex_traits > >::find()'
> boost::regex_traits > >:: construct_init(boost::basic_regex > > const&,

No worries I am actually not a computer engineer either haha. The makefile
actually houses everything you need to recompile voicecommand so once you
have you have deleted that try just running the simple "make"  command
without quotes or you might need to run make -f makefile