StevenHickson/PiAUISuite #52: Voice Command - NO SOUND AT ALL and Weird Symbol

joshuayap94, 2016-01-02:
Dear Githubian,

My PiAUISuite installation will always end up with no audio output. It works fine for aplay but not for voicecommand.

Mind helping me out?

ccurran689, 2016-01-04:
I have the same issue and can't seem to get a response anywhere :(

scooper86, 2016-01-22:
Ditto, still unable to work this out, by all accounts ALSA knows my hdmi monitor w.speakers exists and my USB microphone however it does not get picked up by voicecommand. The microphone seems to be registered when using it however it always comes back with "received improper command".

scooper86, 2016-01-24:
So I managed to get the audio out to work, I amended tts found in /usr/bin or via the source and installing again.

I changed &client=t &client=tw-ob

I also followed the other fix for making the mic output as mono following a fix on here and now speech-recognition.sh works and correctly interprets my speech however running voicecommand still gives me a "received improper command" however now I can hear it too! :P

joshuayap94, 2016-01-24:
So your PIAuiSuite now can received voice and respond back??

scooper86, 2016-01-25:
Yes after a bit of tinkering I have managed to get it working, like I mentioned above changing the voice seemed to be the major hurdle and making sure the recording was sent in mono. After that it was just a matter of tinkering with commands.conf.

joshuayap94, 2016-01-25:
Do you still get weird symbol when you wanna change the RPi keyword?

shrutie, 2016-02-19:
@scooper86 Where did you amend the  tts file??