StevenHickson/PiAUISuite #56: Google topped the TTS service, the PiAUISuite won't work

raqueebh, 2016-01-30:
Hello Steve,

It took 3 days to figure out tts script is not working.
 
Seems like google has blocked that service when its used by automated scripts.
https://support.google.com/websearch/answer/86640

Can you please update like Jasper?

Thanks.

--
Raqueeb Hassan
Bangladesh

CoOral, 2016-01-31:
I've got exactly the same problem, and i'll be thankfull if you might update it.

Best regards

raqueebh, 2016-01-31:
Or you can use offline espeak tts script for rest of us. 

I guess you had that previously.

gymolnargabor, 2016-02-02:
yes, make it offline please... (and if it is not too difficult, can you remove the old videos from youtube, and make new ones for raspbian...it took me a lot of time to figure out, I'm a little bit outdated :) )

shrutie, 2016-02-19:
@raqueebh  How exactly can we use the offline espeak script.
Please mention the change syou made, I am struggling since August.

patrickreidglennon, 2016-03-18:
the tts script seems pretty basic, can't we just drop in another one and wrap it with a simplified /usr/bin/tts ?  Going to give that a try

patrickreidglennon, 2016-03-18:
Changes my tts script to this:

#!/bin/bash                                                                                                                                                               string=$@                                                                            
espeak -ven+f3 -k5 -s150 "$string"  

I can now use tts "this is a test" but I am pretty sure voicecommand is returning null voice recognitions..  Having a look at /usr/bin/speech-recog.sh now, but I'm wondering if we need API keys now...?

krist-jin, 2016-03-27:
I had the same problem but it seems google does block the speech api. I figured it out here: #55 
Hope it helps

irsx02, 2016-04-04:
I have also noticed that google's TTS is no longer working.  I downloaded  pico2wave to replace tts.   

I then wrote a wrapper for tts with this chunk of code 

````
#!/bin/bash
#since google ended TTS, this script replaces tts with pico2wave.

if [ $# -lt 1 ]  
then               #an argument was not entered - pico2wave needs something to say
   /usr/bin/pico2wave -w /tmp/tempsound.wav "I have nothing to say." 
   /usr/bin/aplay /tmp/tempsound.wav
   rm /tmp/tempsound.wav
   exit 0
fi
speech=$@  
/usr/bin/pico2wave -w /tmp/tempsound.wav "$speech"  
/usr/bin/aplay /tmp/tempsound.wav
rm /tmp/tempsound.wav

````
What I found is that if I run tts "say something" it will say something.  Now, when I'm running "voicecommand -c",  I noticed that it will respond based on what's in the command.conf file. So, if i say "who made you" VoiceCommand will reply back with "I was created by Steven Hickson."  However, the first acknowledge response, "Yes Sir" or the i-dont-understand-you prompt, "Received improper command" is not coming out at all. 

I'm looking into /dev/shm/voice.log and such, and I think its something to do with VoiceCommand using   "tts -l en"  (where  -l en for english )  vs pico2wave -l en-US, (where en-US represents US english.)   

So far, I've found speech-recog.sh that has a reference to "tts -l en"    I fixedit to "tts -l en-US"   but I'm still looking for other references.

vaski666, 2016-04-04:
irsx02

where did you place the wrapper?

Colin1964, 2016-04-04:
irsx02

So good to know someone is trying to keep voicecommand alive! Please could you provide a bit more detail on where this wrapper needs to go. Do you just replace the existing tts.sh file with this one?

SweeneyToddTheDemonBarber, 2016-04-04:
Yes! Please keep trying. I've already installed Pico2Wave.

irsx02, 2016-04-05:
Ok, I'm on Raspberry Pi 3, Raspbian (jessie), so your mileage may vary.    I placed my TTS  script at /usr/bin directory.  There should already be a TTS file there. Back it up (you may need to use sudo), copy my tts script, and then make the script executable.  
To test, type in
`tts hello world. my voice is back.
`

And you should hear its greeting.

krist-jin, 2016-04-05:
well seems google tts change its api somehow..
But you can modify the part of call tts api in /usr/bin/tts to fix:
`wget -q -U Mozilla -O output.mp3 "http://translate.google.com/translate_tts?ie=UTF-8&total=1&idx=0&textlen=32&client=tw-ob&q=Test&tl=En-gb"`
This works for me.

irsx02, 2016-04-05:
Update; I made my tts replacement TTS script/wrapper a little more robust - but its now stuck at en-US (US English)   It does solve the issue with the "Yes Sir" and "Receieved Improper Command."  At the expense of behing locked into US English.   Basically, I look for the -l option in tts, and make sure the language is set to "en-US" when it sees "en."
 
````
#!/bin/bash
#since google ended TTS, this wrapper-script replaces tts with pico2wave.
#version 0.2 -now rudimentarily handles language -l param.

if [ $# -lt 1 ]  
then                  #no argument entered - i need something to say
   /usr/bin/pico2wave -w /tmp/tempsound.wav "I have nothing to say." 
   /usr/bin/aplay -q /tmp/tempsound.wav
   rm /tmp/tempsound.wav
   exit 0
fi
 
if [ "$1" = "-l" ]    #-1 in event where user defines language- note:   
then                  # assumes $2 is 'en' or a valid language option. 
   lang=$2
   if [ $lang = "en" ] #cant find the real source, but if i see en 
   then                #i'm hard coding en-US (US english)
      lang="en-US" 
   fi           
   shift 2  
   speech=$@  
   /usr/bin/pico2wave -l $lang -w /tmp/tempsound.wav "$speech" 
   /usr/bin/aplay -q /tmp/tempsound.wav
   rm /tmp/tempsound.wav
   exit 0
else                  #else lets go straight to speech-output
  speech=$@  
  /usr/bin/pico2wave -w /tmp/tempsound.wav "$speech"  
  /usr/bin/aplay -q /tmp/tempsound.wav
  rm /tmp/tempsound.wav
fi 
```` 

I"m going to admit its not ideal, but it is working for me at the moment. It recognizes my keyword, respond appropriately (yes sir!) and awaits my command.  When I tell it to "tell me the time" it replies correctly.

Colin1964, 2016-04-05:
@irsx02 

Firstly, much appreciate the extra detail and also the extra mile with the update.

 I'm working with RpiB with Wheezy (I know I should be on Jessie but I'm being lazy and it has taken me days to get Voicecommand just to do STT). I've installed your replacement TTS script (all of 2 secs, nice'n'easy) but now I am struggling to find a working version of Pico2Wav - when I apt-get libttspico-utils I get package not available?? Searching and looking on other forums suggests I have to compile and build???? Or there is this pico2wave "like" option here - `(https://github.com/ch3ll0v3k/picoPi2)`. If i use this and change refs in your script from "pico2wave" to "PicoPi2Wav" would that work?

Thanks again for responses so far - I'm determined to get this working!!!

Colin1964, 2016-04-05:
So I followed the instructions for PicoPi2Wav and it failed to compile. Looks like I'm going to have to find the real pico2wav.

Colin1964, 2016-04-05:
OK, making some progress - found this guide here [http://rpihome.blogspot.co.uk/2015/02/installing-pico-tts.html](url). Just waiting for it to compile - fingers crossed!!

Colin1964, 2016-04-05:
@irsx02 

SUCCESS!!! Thank You! - She is talking to me again.

saikrishna321, 2016-04-16:
@irsx02 i followed your steps, now i'm able hear back from pi when i type tts "hello i'm back" i can hear the same thing 

when i run the command voicecommand -c i always hear No translation 

can you help me on this ??

```
Opening config file...
You have a formatting error on line 31 of your config file. I'm ignoring that line
running in continuous mode
keyword duration is 2 and duration is 3
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 46331    0    14  100 46317      1   5818  0:00:07  0:00:07 --:--:-- 10476
No translation
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 45918    0    14  100 45904      5  19152  0:00:02  0:00:02 --:--:-- 19158
No translation

```

saikrishna321, 2016-04-17:
@Colin1964 can you paste your tts file in gist that would help me

saikrishna321, 2016-04-18:
@krist-jin i tried the way you have mentioned, still no luck its keeping on saying No translation.

When i say Pie, i get a reply as Yes,Sir but after that when i ask Who made you? or anything it says No translation 

can someone pls help me on this ?

irsx02, 2016-04-19:
@saikrishna321  - Can you go into the PiAUISuite/VoiceCommand directory and look for a shell script called speech-recog.sh  ?    Run the script  `./speech-recog.sh`    and say a few words
If everything goes right,  it should return

```
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 45918    0    14  100 45904      5  19152  0:00:02  0:00:02 --:--:-- 19158
<Whatever you said within 2 seconds>
```

If it still says "no translation", then type  `leafpad speech-recog.sh`  It'll bring up leafpad on the code in speech-recog.sh     Look for near the bottom of the file for "arecord -D $hardware -f ...."    That line is the command that records your voice (arecord),  converts it to flac audo format (flac), send its to google (curl) and parses the results (sed/awk)  

The no translation is happening somewhere within those lines of code.  You can run each section individually to figure out where the "No Translation" is coming from.

irsx02, 2016-04-19:
One more thing. You should see what is the formatting error on line 31 of your config file.   Via a terminal window, from your home directory

`leafpad .command.conf  `

You should see a list of all VoiceCommand acceptable commands.  Jump to line 31 and look for the format error.

saikrishna321, 2016-04-19:
@irsx02 thanks a lot for getting back on it :) i will give it a try and get back to you .

irsx02, 2016-04-19:
@saikrishna321  Also, your issue is more about STT (Speech To Text) and not TTS (Text to Speech) Text to Speech is what pico2wave is trying to solve.  Speech to Text is more about "No translation"." See if anything over in #55 can help you out.  

And remember, if you find the solution, make sure you post it to #55 to help out anyone in the future.

saikrishna321, 2016-04-19:
@irsx02 i did take look at #55 but nothing helped me from that thread..

saikrishna321, 2016-04-19:
@irsx02 

When i ran ./speech-recog.sh i cant hear back anything ..
```
./speech-recog.sh
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 66840    0    14  100 66826      1   8860  0:00:07  0:00:07 --:--:-- 16681
```

when i run voicecommand -c again, it replies Yes Sir. Fixed the error on the .command.conf

after that it says No translation 

```
voicecommand -c
Opening config file...
running in continuous mode
keyword duration is 2 and duration is 3
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 66388    0    14  100 66374      7  35797  0:00:01  0:00:01 --:--:-- 35800
No translation
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 65817    0    14  100 65803      9  46479  0:00:01  0:00:01 --:--:-- 46471
No translation
```

speech-recog.sh 

https://gist.github.com/saikrishna321/833e0206741dbe5f8caff7036d58b090

can you please help me fix this ..

irsx02, 2016-04-19:
The "No translation" error looks exactly like the STT(speech to text) issue mentioned in #55.  Follow @krist-jin instructions from his post.  

From the gist link you provided, I can see the line near the end with `... -f cd -t wav -d ...`    This has been changed and @krist-jin  has described how to fix it.   Google Speech API is not recognizing multi-channel audio anymore.  His instructions will describe how to fix it.

saikrishna321, 2016-04-20:
@irsx02 @krist-jin @ripleyXLR8 

i tried exactly as mentioned by you guys and still no luck .. 

```
#!/bin/bash

hardware="plughw:1,0"
duration="3"
lang="en"
hw_bool=0
dur_bool=0
lang_bool=0
for var in "$@"
do
    if [ "$var" == "-D" ] ; then
        hw_bool=1
    elif [ "$var" == "-d" ] ; then
        dur_bool=1
    elif [ "$var" == "-l" ] ; then
        lang_bool=1
    elif [ $hw_bool == 1 ] ; then
        hw_bool=0
        hardware="$var"
    elif [ $dur_bool == 1 ] ; then
        dur_bool=0
        duration="$var"
    elif [ $lang_bool == 1 ] ; then
        lang_bool=0
        lang="$var"
    else
        echo "Invalid option, valid options are -D for hardware and -d for duration"
    fi
done

#this works really inconsistently and I don't know why. I would love to implement it
#sox -r 16000 -t alsa $hardware /dev/shm/out.flac silence 1 0.3 1% 1 0.5 1%
#wget -q -U "rate=16000" -O - --post-file /dev/shm/out.flac --header="Content-Type: audio/x-flac; rate=16000" "http://www.google.com/speech-api/v1/recognize?lang=en&client=Mozil$
#arecord -D $hardware -f cd -t wav -d $duration -r 16000 | flac - -f --best --sample-rate 16000 -o /dev/shm/out.flac 1>/dev/shm/voice.log 2>/dev/shm/voice.log; wget -O - -o /dev$
#arecord -D $hardware -f cd -t wav -d $duration -r 16000 | flac - -f --best --sample-rate 16000 -o /dev/shm/out.flac 1>/dev/shm/voice.log 2>/dev/shm/voice.log; curl -X POST --da$
arecord -D $hardware -f S16_LE -d $duration -r 16000 | flac - -f --best --sample-rate 16000 -o /dev/shm/out.flac 1>/dev/shm/voice.log 2>/dev/shm/voice.log; curl -X POST --data-b$
rm /dev/shm/out.flac
```

```
inline float GetVolume(string recordHW, string com_duration, bool nullout) {
FILE *cmd;
float vol = 0.0f;
string run = "arecord -D ";
run += recordHW;
run += "-f S16_LE -d";
run += com_duration;
run += " -r 16000 /dev/shm/noise.wav";
if(nullout) run += " 1>>/dev/shm/voice.log 2>>/dev/shm/voice.log";
system(run.c_str());
cmd = popen("sox /dev/shm/noise.wav -n stats -s 16 2>&1 | awk '/^Max\\ level/ {print $3}'","r");
fscanf(cmd,"%f",&vol); fclose(cmd);
return vol;
}
```

sudo apt-get install g++-4.8
make
go to install folder and sudo ./InstallAUISuite.sh

now when i run voicecommand -c cant hear back anything, i had  to change the file /usr/bin/tts as below to head back the voice, i cant hear anything other than Yes Sir?

```
wget -q -U Mozilla -O output.mp3 "http://translate.google.com/translate_tts?ie=UTF-8&total=1&idx=0&textlen=32&client=tw-ob&q=$tmp&tl=En-gb"

wget -q -U Mozilla -O output.mp3 "http://translate.google.com/translate_tts?ie=UTF-8&total=1&idx=0&textlen=32&client=tw-ob&q=$string&tl=En-gb"
```

irsx02, 2016-04-21:
@saikrishna321  first thing, make sure you can record any sound on your mic.    You may search google on how to use arecord / aplay to record and play saved audio.

After you confirmed your sound is working.  Try this

(All lines of code here come from the /PiAUISuite/VoiceCommand/speech-recog.sh file, but I filled in any variables)

First, record your voice and convert to flac 

`arecord -D plughw:1,0 -f S16_LE -d 3 -r 16000 | flac - -f --best --sample-rate 16000 -o /dev/shm/out.flac 1>/dev/shm/voice.log 2>/dev/shm/voice.log`
 
Now, go to /dev/shm directory and make sure there's a out.flac file present.  If its there, run this command;

`curl -X POST --data-binary @/dev/shm/out.flac --user-agent 'Mozilla/5.0' --header 'Content-Type: audio/x-flac; rate=16000;' "https://www.google.com/speech-api/v2/recognize?output=json&lang=en-US&key=AIzaSyBOti4mM-6x9WDnZIjIeyEU21OpBXqWBgw&client=Mozilla/5.0"
`
Look at the result. Its XML (i think)  There should be a parameter called "transcript" followed by 
text of what you said.  

Now, I think this is the STT part that isn't working.  Please show me the results when you run curl.  I'll help you troubleshoot it from there

saikrishna321, 2016-04-21:
@irsx02 will give it a try today, once i have the PI with me :) Thanks a lot for detailed info

saikrishna321, 2016-04-22:
@irsx02 

i can see what i recorded in the XML file 

```
pi@raspberrypi:/dev/shm $ curl -X POST --data-binary @/dev/shm/out.flac --user-agent 'Mozilla/5.0' --header 'Content-Type: audio/x-flac; rate=16000;' "https://www.google.com/speech-api/v2/recognize?output=json&lang=en-US&key=AIzaSyBOti4mM-6x9WDnZIjIeyEU21OpBXqWBgw&client=Mozilla/5.0"
{"result":[]}
{"result":[{"alternative":[{"transcript":"I am from India","confidence":0.77188921},{"transcript":"I'm from India"}],"final":true}],"result_index":0}
```

saikrishna321, 2016-04-22:
@irsx02 ping

irsx02, 2016-04-22:
@saikrishna321, I'm in California, USA.  We're going to play time zone tag.  Plus, I don't have my pi right now, and can't get to testing it until the evening, my timezone.  

 In the mean time, I see from your curl output that its able to contact google correctly.  Curl command  posts your audio file ("I am from India") up to google speech and turn it into text, which is the transcript response you posted.   I'm thinking that  somewhere in your copy of VoiceCommand, its not recording the audio correctly.    I know #55 took care of my issue with "No Translation"    When I get back I can see if I can find more info.

saikrishna321, 2016-04-23:
@irsx02 hope you a great time out there.. 

with further debugging 

I record using the below command 
```
arecord -D plughw:1,0 -f S16_LE -d 3 -r 16000 | flac - -f --best --sample-rate 16000 -o /dev/shm/out.flac 1>/dev/shm/voice.log 2>/dev/shm/voice.log
````

Now i curl and i see XML response 

```
pi@raspberrypi:/dev/shm $ curl -X POST --data-binary @/dev/shm/out.flac --user-agent 'Mozilla/5.0' --header 'Content-Type: audio/x-flac; rate=16000;' "https://www.google.com/speech-api/v2/recognize?output=json&lang=en-US&key=AIzaSyBOti4mM-6x9WDnZIjIeyEU21OpBXqWBgw&client=Mozilla/5.0"
{"result":[]}
{"result":[{"alternative":[{"transcript":"I am from India","confidence":0.77188921},{"transcript":"I'm from India"}],"final":true}],"result_index":0}

```

as we recorded the above response i tried to hear it back with aplay /dev/shm/out.flac i hear a old tv sound nothing other than that .. 
```
Playing raw data '/dev/shm/out.flac' : Unsigned 8 bit, Rate 8000 Hz, Mono

```

saikrishna321, 2016-04-23:
@irsx02 hope you had a great time out there.

with further debugging 

when i run speech-recg.sh 

```
speech-recog.sh
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 86900    0   140  100 86760     68  42183  0:00:02  0:00:02 --:--:-- 42198
"hello how are you doing"
```
When i did a ./InstallAUISuite.sh, during the setup i was able to hear back on the setup check 
but 

Now when i run the voicecommand -c 
```
voicecommand -c
Opening config file...
running in continuous mode
keyword duration is 2 and duration is 3
```

just hangs at this place ..

irsx02, 2016-04-25:
@saikrishna321   I noticed that on my Raspberry Pi 3, when I run aplay /dev/shm/out.flac, it only returns static - also like the sound of an old TV.  I assume its because aplay can not handle flac file format.  

(Can any experts verify this? Can aplay understand the flac audio format? ) 

The good news is that you can get speech-recog.sh to work. Based on your output, it is changing your recorded voice into text.   Which is great!

I'm not sure what's up with the voicecommand -c right now - I'll keep researching on it.  Do you see any error messages showing up in the  /dev/shm/voice.log file?  (` leafpad /dev/shm/voice.log `)

As a side note, each time your install piAUISuite/VoiceCommand, you will have to fix the tts script.  One way around it is that I've found if you directly edit the tts in the /PiAUISuite/VoiceCommand directory, it will install with the fixed tts every time.

saikrishna321, 2016-04-25:
@irsx02 

I will check the voice.log today and will keep you posted. When i changed the bit Rate to 48000 from 16000 and then aplay was able to play the audio i recorded. But why voicecommand -c does keeps on researching not sure...

saikrishna321, 2016-04-25:
@irsx02  before are the logs... I see this even during the setup... 

```
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
ALSA lib conf.c:4617:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:4722:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument
```

saikrishna321, 2016-04-25:
@irsx02 
After running the Update.sh, running the speech-rech.sh i can still the text and tts Hello talks back... 

the log looks like this ..  

running voicecommand -c, i dont hear back Yes Sir!

```
pi@raspberrypi:~/Documents/PiAUISuite/VoiceCommand $ voicecommand
Opening config file...
running in continuous mode
verifying command as well
keyword duration is 2 and duration is 2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 99702    0    14  100 99688      5  40505  0:00:02  0:00:02 --:--:-- 40523
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 88735    0    14  100 88721      7  49235  0:00:01  0:00:01 --:--:-- 49234
```
```
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

flac 1.3.0, Copyright (C) 2000-2009, 2011-2013  Josh Coalson & Xiph.Org Foundation
flac comes with ABSOLUTELY NO WARRANTY.  This is free software, and you are
welcome to redistribute it under certain conditions.  Type `flac' for details.

noise.wav: 73% complete, ratio=0.578^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^Hwrote 102228 bytes, ratio=0.532
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

flac 1.3.0, Copyright (C) 2000-2009, 2011-2013  Josh Coalson & Xiph.Org Foundation
flac comes with ABSOLUTELY NO WARRANTY.  This is free software, and you are
welcome to redistribute it under certain conditions.  Type `flac' for details.

noise.wav: 73% complete, ratio=0.498^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^Hwrote 93399 bytes, ratio=0.486
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

flac 1.3.0, Copyright (C) 2000-2009, 2011-2013  Josh Coalson & Xiph.Org Foundation
flac comes with ABSOLUTELY NO WARRANTY.  This is free software, and you are
welcome to redistribute it under certain conditions.  Type `flac' for details.

noise.wav: 73% complete, ratio=0.478^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^Hwrote 87720 bytes, ratio=0.457
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

irsx02, 2016-04-25:
@saikrishna321  
  I noticed that in your voice.log,  you are seeing  
```
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM plughw:1,0-f
arecord: main:722: audio open error: Invalid argument  
```
This means somewhere out in the code, there is a missing blank space between the device "plughw:1,0" and the parameter " -f"    You will have to look exactly where in the code, but its suppose to say  "plughw:1,0 -f "   but instead its saying "plughw:1,0-f"   (that lack of space between the 1,0 and -f is messing with the audio.

irsx02, 2016-04-25:
One more thing to try out.  In the voicecommand.cpp file, under the GetVolume(string recordHW, string com_duration, bool nullout)  function, look for the line

`  run += "-f S16_LE -d";   `

I think its suppose to have a space between the " and the - f   So it should look like this

`  run += " -f S16_LE -d ";   `

You will then need to run `make` (to compile the C++ code),  install (AUISuite) and fix tts.sh again.  Test this out and see if its the case.

saikrishna321, 2016-04-26:
@irsx02 i did that and when i run voicecommand -c i see this below but dont hear anything 

```
voicecommand
Opening config file...
running in continuous mode
verifying command as well
keyword duration is 2 and duration is 2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 82884    0    14  100 82870      8  48294  0:00:01  0:00:01 --:--:-- 48292
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 80873    0    14  100 80859      1  11059  0:00:07  0:00:07 --:--:-- 16930
```

voice.logs

```
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

flac 1.3.0, Copyright (C) 2000-2009, 2011-2013  Josh Coalson & Xiph.Org Foundation
flac comes with ABSOLUTELY NO WARRANTY.  This is free software, and you are
welcome to redistribute it under certain conditions.  Type `flac' for details.

noise.wav: 73% complete, ratio=0.451^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^Hwrote 82870 bytes, ratio=0.432
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

flac 1.3.0, Copyright (C) 2000-2009, 2011-2013  Josh Coalson & Xiph.Org Foundation
flac comes with ABSOLUTELY NO WARRANTY.  This is free software, and you are
welcome to redistribute it under certain conditions.  Type `flac' for details.

noise.wav: 73% complete, ratio=0.423^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^H^Hwrote 80859 bytes, ratio=0.421
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE '/dev/shm/noise.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

irsx02, 2016-04-27:
@saikrishna321   I'm lost as to how to figure out this puzzle.  I'm going to recommend delete  and reinstall.   Re-download all the files from github.  Before running install, make the following changes (which I am using memory, so please do double-check first)
1. update PiAUISuite/VoiceCommand/tts  with the code mentioned earlier in this post 
2. update PiAUISuite/VoiceCommand/speech-recog.sh old line `-f cd -t wav`  with ` -f S16_LE -d`   as described in  ripleyXLR8  post in #55. 
3.  update PiAUISuite/VoiceCommand/voicecommand.cpp function GetVolume(string RecordHW,...) Again, look for `-f cd -t wav`  with ` -f S16_LE -d`  This again is described in more detail on #55.
4. install G++4.8 (which I'm sure you have done so already) run make. This is described by krist-jin in #55.
5.  Now, install VoiceCommand
6. Testing time. a) First, confirm your mic works (arecord/aplay)  
                          b) run tts "your message here"    - Speaker should say "your message here"
                          c) run speech-recog.sh   - whatever you say it two seconds will be output in text on screen..
                          d) run voicecommand -c

saikrishna321, 2016-04-27:
@irsx02 Finally 👍 got it worked :) I followed the steps as you suggested, i had to change the channel to 32000 rate and it did work :) i can hear back ... 

An other question, sorry --- Is there a way we can check google and get back for responses like Ok Google ?

irsx02, 2016-04-28:
That's awesome - looks like a reinstall of VoiceCommand was all it took to get it to work.

As for the google / Ok google feature - I've been searching for something like that idea, but I have not found anything.   The closest I did get (and by close, like India to California close)  was to use the google function in VoiceCommand.  I had to alter the google.sh script to use iceweasel to do a search.  (I don't have midori)  

Basically, I went to /usr/bin/google.sh and changed the code in it to call iceweasal.   

Now, when I say "Google What is Raspberry Pi " VoiceCommand will open up a new tab in Ice Weasal witha search of "what is raspberry pi"    No, its not exactly like "OK Google." But it is something.

saikrishna321, 2016-04-28:
@irsx02 Will give it a try ... Take a look at this 

https://oscarliang.com/raspberry-pi-voice-recognition-works-like-siri/

irsx02, 2016-04-29:
@saikrishna321    Based on what I've read on this site ([ http://diyhacking.com/best-voice-recognition-software-for-raspberry-pi/])  and on Oscar Liang's site,  both Oscar's and Voice Command 3.0 are very similar - uses google for stt, and google API for tts (although I had to change it over to Pico2Wave, which is owned by Google) 

If you're familar with python, you may perfer Oscar's Voice Recoginition.  

But so far I like how I am able to use Voice Command's command.conf file to write my own commands and what to execute.  And I'm a little bit familar with C++ 
 
EIther way, neither of these two options have an "OK google"-like feature. That's what I am looking for.

saikrishna321, 2016-04-29:
@irsx02 when you say neither of them have a feature like OK google-- i thought Oscar's Voice Recoginition has it right... let say when you ask OK Google-- What is 3 times 3 it return you 9, its the same with Oscar's Voice Recoginition. But with DIYHacking it just pulls from the command.conf and in return would need as  execution statement .. 

Just correct me if i'm wrong out there ?

saikrishna321, 2016-04-29:
@irsx02 i see something weird, i notice once i run the voicecommand -c and ask for the first time who made you ;; it replies the right one.. Now it keeps on saying yes sir and waits for command.. i was expecting Yes Sir only when i call the keyword 

```
#This is the default config file
#These are the special options you can set (remove the #)
#!verify==1
!keyword==Sam
#!thresh==0.7
!continuous==1
!response==Yes Sir?
#!quiet==0
#!ignore==0
!filler==0
!duration==2
#!com_dur==3
#!hardware==plughw:1,0
#Here are the commands
show me==/home/pi/AUI/Imaging/test 2
track me==/home/pi/AUI/Imaging/test 1
download==download ...
play $1 season $2 episode $3==playvideo -s $2 -e $3 $1
download $1 season $2 episode $3==download $1 s$2e$3
play==playvideo -r -f ...
multiple==playvideo -r -m -c 5 ...
download==download ...
YouTube==youtube-search ...
Google==google ...
~music==xterm -e pianobar
~weather==python /home/pi/Documents/pisuite/weather.py
~made you==tts "I was created by Steven Hickson" 2>/dev/null
~music==xterm -e control-pianobar.sh play
~terminal==xterm &
```

SweeneyToddTheDemonBarber, 2016-04-29:
Yes my software is having the same problem. Everything works perfectly except the keyword has 0 confidence. Any loud noise activates my keyword (which is Perkins), even clapping. I've already done what's written here and in #55 What can I do?

saikrishna321, 2016-05-01:
@SweeneyToddTheDemonBarber when you try ./speech-recong.sh say something now.. and see if it prints completely what  u say

HangLoooose, 2016-05-01:
I followed all instructions given by @krist-jin
Now speech-recog.sh kann translate my voice into the correct text and tts says whatever you want it to say. But as soon as I start voicecommand and say my keyword the following lines appear:

Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 25258 0 14 100 25244 9 16760 0:00:01 0:00:01 --:--:-- 16762
No translation

I changed the sample rate as @saikrishna321 suggested but it still says "No translation".

Can anyone help me to fix this issue?

irsx02, 2016-05-01:
@saikrishna321  - Ok, your issue is a unique one that I haven't seen before.  I'm wondering, after it  says "yes sir" and waits for command, does it take a new command? 

@SweeneyToddTheDemonBarber - Just as SaiKristna321 states, what was the result when you run speech-recog.sh?   What is the one letter you mention?  And does the "who made you?" command work in your situation? (does it respond with "I was created by Steven Hickson?")

@HangLoooose   - I'm betting on the "No translation" issue is coming somewhere in the VoiceCommand.cpp file.  Did you make changes to the code?  Spaces can affect what is sent to google.  And, if all else fails, you can try reinstalling the whole thing.

saikrishna321, 2016-05-02:
@irsx02 ya it does take new command.. it was just the first time i called the keyword. Post that any sound like hand clap also it responds as Yes Sir!, has this got something to do with threshold

```
voicecommand -c
Opening config file...
running in continuous mode
keyword duration is 2 and duration is 2
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 62769    0   189  100 62580    113  37609  0:00:01  0:00:01 --:--:-- 37608
command: tts "I was created by Steven Hickson" 2>/dev/null
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 57452    0    14  100 57438      9  38577  0:00:01  0:00:01 --:--:-- 38574
No translation
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 58153    0    14  100 58139      9  40092  0:00:01  0:00:01 --:--:-- 40123
No translation
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 32000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
```

Bokaj707, 2016-05-03:
@saikrishna321 Make sure your config file has verify set to 1 and it isn't commented out.
`!verify==1`
This insures the keyword is found before asking for a command. When you start voicecommand, you should see the line 
`verifying command as well`

That being said, I haven't been able to get it to recognize any keyword when it's verifying. speech-recong.sh shows it's recording, but I never get a `Found audio` trigger.

saikrishna321, 2016-05-08:
@Bokaj707 when i have the !verify==1 i dont hear the command Yes Sir!, once that is commented i hear it back ..

SweeneyToddTheDemonBarber, 2016-05-13:
Hey guys @saikrishna321 @irsx02 , thanks for your responses. I got everything working with pico2wave, having changed tts. ./speechrecognition.sh works perfectly. The problem I have is that any sound triggers the keyword. I've used words with different syllables (current one: Hello Perkins) and yet if I clap it stills triggers the keyword. :(

As for the One letter, I'm trying to add my Spotify music (through mpd) and when I search for an artist it only looks for the first letter of their names. I started looking into it so I made a new command:
`repeat $1==tts $1`
This only says the first letter of the word/phrase. I started adding more variables ($2...$5) and they keep spelling the word I''ve said.
Example:
`repeat $1==tts $1`
I say: Repeat What day is it
     the program answers: W
`repeat $1$2$3==tts $1$2$3`
I say: Repeat What day is it
     the program answers: W H A

But my main concern is the first one, where anything triggers the keyword.
Thanks!

SweeneyToddTheDemonBarber, 2016-05-13:
Guys we DID IT!!!!
After many tries we finally did it. The trick was that I had to make the file after editing voicecommand.cpp. It's been working great. As for my second problem I found a solution. you have to put the $1 before a command and it works.
Thanks

saikrishna321, 2016-05-14:
@SweeneyToddTheDemonBarber can you please post your tts file and how does your command look when its $1 ? Does this solve the trigger of voicecommand for any keyword?

[25 more replies]