StevenHickson/PiAUISuite #58: Dependency issue libboost-regex 1.49.0 when installing on OSMC

vaski666, 2016-04-01:
Hi,

Not sure if this the right place to ask.
When trying configure voice command it says that  libboost-regex1.49.0 is missing.
Trying to install this package gives me the following : Couldn't find any package by regex 'libboost-regex1.49.0'

This issue has been posted on various sites by different users, but no one has posted a solution or if it is impossible to do so.

-John

vaski666, 2016-04-01:
Update: have created a symlink using the following command.
ln -s /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.55.0 /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.49.0

this has fixed the problem

-john

closed 2016-04-01 by vaski666

Dishcreate, 2017-03-15:
Hii I had that issue when I configure voicecommand in raspberry pi 3. But It was fix by installing libboost_regex.so.1.49.0

borinbilly, 2017-03-25:
Failed to create symbolic link: File exists

enannos, 2017-04-12:
First: 
```bash
sudo apt-get install libboost1.49-dev 
```
Then install the regex:
```
sudo apt-get install libboost-regex1.49-dev
```
Install dependencies? **NO**
That did the trick for me.
Evangelos

technotodd, 2017-11-20:
So this is still an issue. Instead of installing 1.49-dev (which is no longer available) I installed libboost.1.62-tools-dev, then tried running voicecommand -e and still got the same error "cannot open shared object file: no such file or directory"

[My ultimate goal here is to finish the http://www.instructables.com/id/Wireless-Multi-Channel-Voice-Controlled-Electrical/ project using a Raspberry Pi 2 running Raspbian Stretch with Desktop.

Any help would be greatly appreciated. Oh the guy, Steve Hickson, who wrote the voicecommand code also added an image that was supposed to fix this and I was unable to get the image to load on my RPi

mrFroggy, 2017-12-04:
technotodd
cd cd /usr/lib/arm-linux-gnueabihf/
sudo ln -s libboost_regex.so libboost_regex.so.1.49.0

kevanmartin, 2021-01-31:
I have just installed Steven Hickson's Raspberry Pi Voice Control software on a Pi3. All goes well until I enter voicecommand -s. The following error comes up.
voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory
I have tried several ways to reintstall libboost but always get the same error. Has this error been sorted?