StevenHickson/PiAUISuite #61: Can someone supply a Raspberry Pi 2 Img file?

scoobyd00, 2016-05-15:
Im looking at installing this on my raspberry pi 2 to do some home automation voice control but looking at some of the issues it looks like google tts doesnt work anymore and this has been fixed using pico2wave.
Im unsure how to compile the code etc etc to get this latest version to run.

Could someone supply a sd img file from a current working setup they have?
Thanks.

scoobyd00, 2016-05-17:
Took me a few hours today but i managed to pull from github and get it 99% working

closed 2016-05-17 by scoobyd00

johnsonld123, 2016-05-17:
@scoobyd00 can you post your image or details? I am in the same boat as you after trying all day yesterday. I was able to get sound but now it  seems I need to start from scratch. After so many changes I am not sure what to try next.

scoobyd00, 2016-05-17:
I basically did this:-

sudo apt-get install -y libboost-dev libboost-regex-dev youtube-dl axel curl xterm libcurl4-gnutls-dev mpg123 flac sox

git clone https://github.com/StevenHickson/PiAUISuite.git

I then used this guide to install pico2wave

http://rpihome.blogspot.co.uk/2015/02/installing-pico-tts.html
If it comes up with some errors that libs are missing, just apt-get install the ones it needs when it throws the error. It only threw one error for me.

After this

Implement the TTS solution found in #56  to fix the speech output
Then implement #55 to fix the speech recon

This worked for me and its a working solution.

If you need the 8gb img file let me know where i can upload it to. It will be friday until i can do this due to the thresholds imposed by my ISP.

digiltd, 2016-06-21:
@scoobyd00 I am not requesting the image and I only read the thread as was linked from another

But it is good to know that if you Zip the image it will squeeze out all the empty space and considerably reduce the size. It is a simple way to take a working backup of a working configuration in case you have to revert or the SD card dies

e.g my Alexa image was 7.97gb, zipped it is 1.53gb