StevenHickson/PiAUISuite #63: Voicecommand doesn't work with verifying the keyword

blackbelt96, 2016-05-22:
I have everything working finally, but only with verify the keyword set to 0. If I set it to verify the keyword (I'm using Jarvis, but I have tried other keywords too) it doesn't detect that I ever say it. In setup it hears "Jarvis" loud and clear when I change the keyword. Speech-recog works perfectly otherwise, as does tts. 

Also it always says 14 under received when verify keyword is set to 1, no matter what if anything is said.  It also always says 5 average download every time.

joebotbr, 2016-07-31:
Exactly the same here. Everything working, but if verify keyword is set to 1 it's not recognised. If set to 0, commands works as expect. Any solution @StevenHickson ?

MisterD4N, 2016-08-02:
I have the exact same problem here as well. Any help would be appreciated.

blackbelt96, 2016-08-02:
Sorry guys, I spent another few weeks after posting this doing everything I could think of. Uninstalled and reinstalled PiAUISuite and the entire OS over and over again. Countless other things I have forgotten I did by now. Eventually gave up so unless @StevenHickson can shed some light it seems like a dead end as far as I know. 

I do wonder, are you guys using Pi 3s? I don't know WHY it would cause this, but any thing is possible, and it seems Steven has been busy with other things since before the 3 was released.

joebotbr, 2016-08-02:
I'm using RPi 2

MisterD4N, 2016-08-02:
I'm using a RPi 3 here. I fixed the tts problems with Pico and Speech-recog works correcty but voicecommand will not answer to a keyword...

MisterD4N, 2016-08-02:
I've found an option to do what I want. Figured it might be of use for others as well. A big plus is that it is entirely offline.

https://snowboy.kitt.ai/

joebotbr, 2016-08-03:
In voicecommand.cpp we have these lines:
`
//printf("message: %s, keyword: %s\n", message, vc.keyword.c_str());
     if(iequals(message,vc.keyword.c_str())) {
          message[0] = '\0'; //this will clear the first bit
          ProcessVoice(cmd,vc,message);
     }
`
I have uncommented the first line and recompile, but the output is always:

message:, keyword: ***my keyword***

This line compares "message" to "keyword":

`if(iequals(message,vc.keyword.c_str())) `

joebotbr, 2016-08-04:
I've made several changes in voicecommand.cpp but finally I've make it works with keyword recognition.

joebotbr, 2016-08-04:
I'll put all changes here next days ;)

blackbelt96, 2016-08-05:
@MisterD4N Looks interesting. Might have to give it a shot, thanks for letting me know about it!

@joebotbr I cant wait for the write up! what was actually causing it to not work?

joebotbr, 2016-08-05:
@blackbelt96 I can't figure out what was wrong. I've tried to debug it, but the response was always blank, so I used parts of "voicecommand -s" that do exactly the same and was working.

Colin1964, 2016-09-12:
Try adjusting the Threshold value in the .commands.conf file. I found that with verify set to "1" the setting of the microphone threshold is critical. 
My theory is: 

If set too low
-> Speech.recog output is garbage i.e. the STT conversion is corrupted 
-> keyword not recognised 
-> Voicecommand goes back to "listening" mode 
-> no response shown😫

if set too high 
->  STT conversion is not activated 
-> No keyword has been spoken 
-> Voicecommand remains in "listening" mode 
-> no response is shown. 😤

With verify off, all and any sounds heard are processed so there is a good chance that you will get a response especially if only one word e.g. "made" needs to be recognised  as in "who made you". Worth a try! I have verify set and it took a bit of tweaking but "Yes Sir" is working for me 100% in response to my keyword "Gladys" on RPi3 with latest Jesse distribution.

Woody741, 2016-09-21:
I m realy bored with this lib, I have too much problems too, you use cpp code to use a shell script, why you don't use only cpp code, I have taken a look at the shell code, it is not difficult to convert it.

Colin1964, 2016-09-26:
@blackbelt96 @joebotbr @MisterD4N Appreciate posts on this thread are getting a bit old now but if you still have issues thought you might want to try suggestion on #76 that got keyword recognition working. Basically you need to be running in continuous mode  i.e. 

`!continuous==1 `

Looks like keyword verification may not be fully implemented in non continuous mode - @StevenHickson any thoughts?.

bjacqu03, 2016-10-03:
I am running in continuous==1 and I still can not get the keyword recognition working. 

TTS works
if I run in non continuous mode my commands all work, but nothing happens when I run in continuous mode and speak the keyword infact I can't even speak my commands. This is on Pi3.

Has anyone had any luck fixing this?

Colin1964, 2016-10-03:
@bjacqu03 can you post the screen output you get when you first run voicecommand with `continuous==1` and then with `continuous==0` For both cases say your keyword. Also can you post the contents of your .commands.conf file.

You say your "commands" are working does this mean you can confirm that speech_recog.sh is working and recognising your voice? Do you see your spoken words on the display and "Found audio"?

blackbelt96, 2016-10-06:
@joebotbr @MisterD4N @Colin1964 @Woody741 I had given up on this project but I just discovered this [https://github.com/alexa/alexa-avs-sample-app](url) which I will probably be setting up over the weekend. It is being actively supported even by amazon themselves, I would highly recommend you guys check it out if you are still interested in a voice controlled project but still cant get PiAUISuite to work.