StevenHickson/PiAUISuite #64: Trouble w/PiAUISuite

wildbi111, 2016-05-22:
I have a problem with PiAUISuite.  I installed via the directions and the 1st issue is voicecommand would not run as it could not find /usr/lib/libboost_regex.so.1.49.0 so I created a sym-link to libboost_regex.so.1.50.0 and it ran seemingly ok after that.  But I cannot get it to recognize any voice commands.  I am running it as root and have copied /home/commands.conf to /root/.commands.conf.  I have also verified the mic work using arecord and aplay.  arecord used plughw:1,0 when it worked so I modified the .commands.conf to reflect !hardware==plughw:1,0

But thus far no success.  Any thoughts ideas or solutions would be appreciated.

wb

wildbi111, 2016-05-22:
Some additional troubleshooting as I have the entry "hello==echo Hello" in the .command.conf file
so using -I seems to work.

# voicecommand -I Hello
Opening config file...
command: echo Hello
Hello

wildbi111, 2016-05-24:
Just curious but why is this issue/help request closed?

blackbelt96, 2016-05-26:
Are you using a keyword? If you are try running it without a keyword and see if that works. I had the same problems (and then some) and that worked for me. Still can't get it to work with a keyword though.

wildbi111, 2016-05-29:
It currently works with the **~** commands but the keyword never worked.  
(most of my comments are in the issue **56** thread)