StevenHickson/PiAUISuite #67: Waiting for keyword offline?

andreaskoelsch, 2016-06-06:
Hi,
Is it possible to wait for the keyword offline and then do the actual speech-to-text online? I want to have this running all the time, but I do not want all my conversations being uploaded to google, lol. It would be cool to have some offline recognition for the keyword like "Pi". When it recognizes the keyword, it says "Yes, Sir" and the following speech goes to google... Is this, how it is done, or is everything uploaded to google to get text out of it?
Thanks

scoobyd00, 2016-06-07:
I don't think this is possible but would be great if it was. I run mine constantly and it picks up every word off the tv. I set my threshold high to stop this but it means I have to shout at the pi now!

If this was possible it would be a great addition.

scoobyd00, 2016-06-07:
Another option would be to use a hardware button ..so you can just press a button instead of using fhe keyword.

irsx02, 2016-06-14:
I've been loosely researching ways to improve VoiceCommand - I had the same idea as @andreaskoelsch  and have another local program to look for the keyword.  This was due to the way VoiceCommand would record, then send the sounds to google for translation.  I found that rather slow.  I was thinking about having another program do the first round of listening, just for the keyword. 

Except I haven't found any good replacement - my opinion is that  I think google is, so far, the best speech-to-text out  there for the Pi.

Option 2 was using a hardward button - Hardware Override Pushbutton Switch.  Where I put HOPS=true in the command.conf file to turn on the override on the keyword.  Except I would have to learn how to program C++ to do it.  Which, I don't have that much free time to do that approach.  
 
Who knows, if I have some more time I'll see what I can do.