StevenHickson/PiAUISuite #72: Github code outdated

Zulex, 2016-06-27:
Hello all,

I started a week ago with testing PiAUISuite. The main code given by this github page is full with errors. I tried fixing them by reading many topics but can't seem to get it working.

The sound output doesn't work with my usb microphone/headset (i'm using the Blue Yeti) while arecord an aplay seem to work fine. One of the errors i'm getting is 'channels count non available'. 
I replaced the original tts with pico2wave. 

Is Steven still updating the main code or is there someone willing ( with a working version ) to upload it all together for easy install?

Thanks

StevenHickson, 2016-06-27:
Due to Google's frequent changing of the way their speech recognition and tts work (they haven't wanted us to be creating and using an API to those services and now they want to charge for that service), the code base has to change frequently to keep up with it.

Because of this, there are many times when the code base is not working.
Its currently been in this state for a while because I've been using Amazon's alexa service instead. At some point in the near future, I'm going to switch this to a different speech recognition and tts engine that has a bit more support. I've been testing with a couple.

You can test with a couple of the PicoWave PRs that have a working version.
https://github.com/StevenHickson/PiAUISuite/pull/59

closed 2016-06-27 by StevenHickson

diegodeodati, 2017-01-22:
Audio wont work... have you fix google tts ?