StevenHickson/PiAUISuite #74: Voicecommand -c doesn't work

nparabz, 2016-09-10:
Running it on Raspberry Pi 3. When I run voicecommand (not continuous) it works fine, but when I run voicecommand -c it just shows the following screen and doesn’t do anything:

Opening config file…
Not querying for answers
running in continuous mode
keyword duration is 5 and duration is 5

Please help solve this issue

apc00008, 2016-10-06:
same problem. using pico2wave for tts because the old command is not working anymore. the app works only when user type 'voicecommand' in terminal.

Wikibear, 2016-10-23:
Same here. I don't get any answer. Only voicecommand itself works, but also without any verify.

matthewvwalker, 2016-12-11:
I have the same problem.  It did work for a bit, then stopped.

Another issue I see that may be related is after I run "voicecommand -i" and get and improper response it hangs until I break out with ctrl+c .  Possibly the two issues are related?

yannwrck, 2017-01-26:
I have the same problem. Also Raspberry Pi 3. Would be great if someone can fix this.

matthewvwalker, 2017-01-26:
I ended up just switching to snowboy.  Worked perfectly out of the box and is offline.