StevenHickson/PiAUISuite #75: Failed to connect to www.google.com port 443: Connection timed out

KiAlexander, 2016-09-12:
I am a Chinese student and I am confronted with the following problem looking foward to any good advice:
pi@raspberrypi:~$ voicecommand
Opening config file...
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:06 --:--:--     0curl: (7) Failed to connect to www.google.com port 443: Connection timed out
No translation

KiAlexander, 2016-09-12:
what can I do with the problem?I have stuck for a long time.

apc00008, 2016-10-07:
no internet connection or slow internet connection

BHEADRICK, 2017-01-09:
google is blocking automated api calls to the tts service. 

see my comment on this issue
https://github.com/StevenHickson/PiAUISuite/issues/80