StevenHickson/PiAUISuite #76: Keyword is recognized, but as a query

mathijsvandenhurk, 2016-09-22:
Hello there, after putting in some effort I got this to work on my rpi3. It recognizes spoken queries and gives me back answers in speech also. The only thing not working is the verification keyword. I've set it as "hi" or "hello" and it recognizes that I say those words, but instead of receiving the response, it gives me "Attempting to answer: hello" when ignore is off and "Received improper command: hello" when ignore is on. Does anybody recognize this and/or knows how to fix this?

Colin1964, 2016-09-23:
Your issue seems to be more related to the !Verify command rather than !Ignore. Do you get the same output with !Verify set to 1? I think you may be confusing the Ignore command and Verify command. The Ignore command relates to the specific commands you list in the .commands.conf file. If Ignore = 1 then Pi will only search for and recognise commands listed in the config file. If you say "Hello" and you don't have a command called "hello" listed and Ignore is set to 1  the response will be "improper command". With Ignore set to 0 Pi will try and answer any command it hears not just those listed in config. It can only do this though if you have set up a search provider like Wolfram Alpha api which Pi uses to find an answer. Without the api set up it won't find an answer......it will just attempt to. AFAIK Voicecommand was originally linked to a Google api to retrieve answers but along with the TTS api, changes by Google have stopped this working. So if you want answers to general questions like "where is the Shard" you need to link the Wolfram api. If your Pi is responding to everything it hears then you probably have verify = 0. With this set to 0 you disable the verification process so even if it correctly hears and translates the keyword it will not recognise it as a keyword (because verification is off) and not issue the keyword response - instead it is trying to find a command or a general response to what it heard.

Try adding this to your .commands.conf file:

~hello==tts "this is a command not a recognised keyword"

Run the VC and say hello. If you get a response then you need to look closer at your verification setup.

mathijsvandenhurk, 2016-09-23:
Thanks for your reply, I understand that the problem lies in the verify command, I just put the ignore output there for clarification (which obviously didn't have the desired effect :p). I have !verify==1 set in my config file, so this also isn't the problem. It seems that the config file just doesn't recognize the verify setting. I think this might be related to the issues mentioned in #56 I did some changes in the voicecommand.ccp file, maybe I have to rune make, or have a closer look at the language syntax in there. I will post when I have tried this. If anyone has any other suggestions in the meantime, I'm all ears :)

Colin1964, 2016-09-23:
Apologies. I wasn't sure if you were clear on the !Verify command - clearly you are!

Do you get any other output displayed? On my system I get a readout of several parameters whenever a command (or keyword) is issued - something like this:

keyword duration is 2 and duration is 3
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 46331    0    14  100 46317      1   5818  0:00:07  0:00:07 --:--:-- 10476

Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 45918    0    14  100 45904      5  19152  0:00:02  0:00:02 --:--:-- 19158

Do you see the above displayed before your "Attempting to answer" or "Received Improper command" is displayed?

Colin1964, 2016-09-23:
BTW (and again, apologies if I am telling you what you already know), if you changed the .cpp file you do need to recompile it. See comment from RipleyXLR8 and Krist-JN in #55

mathijsvandenhurk, 2016-09-23:
Thnx again for your reply and no apologies needed, when I reread what I wrote I understand why you came to that conclusion :) 

Yes, I get the whole found audio response and then followed by "Attempting to answer" or "Received improper command". Today and tonight I have very little time, but tommorow I wil start with recompiling the cpp  file I changed and post my findings back here, thnx again!

Colin1964, 2016-09-23:
@mathijsvandenhurk Sounds good and hope you have success. I spent several, very frustrating weeks trying to get this working on my RPI3. Now it is working perfectly and I'm just so keen to help others avoid those weeks of frustration if I can!!

mathijsvandenhurk, 2016-09-24:
@Colin1964 Thnx, but no succes so far. I did the following (as shown in issue #56 ):
sudo apt-get install g++-4.8
make
And then reinstall the PiAUISuite

Still I get the very same result... While I can give commands like "who made you" and get the answer back. The keyword is still not recognized. I've noticed that it also isn't outputting "verifying command as well" or "found audio" when I run voicecommand or voicecommand -v, I just get this:

"pi@raspberrypi:~/PiAUISuite/VoiceCommand $ voicecommand 
Opening config file...
Not querying for answers
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 42039    0   214  100 41825    128  25035  0:00:01  0:00:01 --:--:-- 25029
Received improper command: hello"

This is what my config file looks like, maybe I'm missing something:

"#This is the default config file
#These are the special options you can set (remove the #)
!verify==1
!keyword==hello
!continuous==0
!response==Yes Sir?
!quiet==0
!ignore==1
!duration==2
!com_dur==3
!hardware==plughw:1,0
!lang==en
!filler==0
!threshold==8
#Here are the commands
show me==/home/pi/AUI/Imaging/test 2
track me==/home/pi/AUI/Imaging/test 1
download==download ...
play $1 season $2 episode $3==playvideo -s $2 -e $3 $1
download $1 season $2 episode $3==download $1 s$2e$3
play==playvideo -r -f ...
multiple==playvideo -r -m -c 5 ...
download==download ...
YouTube==youtube-search ...
Google==google ...
~music==xterm -e pianobar
~weather==/home/pi/AUI/Misc/sayweather.sh
~made you==tts "I was created by Steven Hickson" 2>/dev/null
~music==xterm -e control-pianobar.sh play
~terminal==xterm &
~Internet==midori &"

Any help would be more than welcome!

Colin1964, 2016-09-24:
@mathijsvandenhurk Have you tried setting !continuous==1   ?

I have been running in continuous mode but did a quick test setting continuous to 0 and got a similar result to your issue:

pi@raspberrypi:~/PiAUISuite/VoiceCommand $ voicecommand
Opening config file...
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 28647    0   221  100 28426     56   7287  0:00:03  0:00:03 --:--:--  7286
Attempting to answer: Gladys
Input interpretation : Gladys  (female given name)
Information for US births : rank | 2233rd
fraction | 1 in 21317 people (0.0047%)
number | 83 people per year
(US data based on 2015 births and other SSA registrations in the US)
Estimates for current US population : expected total number alive today | 90776 people

As you can see, my keyword "Gladys" was recognised but not as a keyword but as a search term that Wolfram then went on to answer. Also I didn't get the keyword response of "Yes Sir" either.

I reset Continuous back to "1" and everything went back to normal:

pi@raspberrypi:~/PiAUISuite/VoiceCommand $ voicecommand
Opening config file...
running in continuous mode
verifying command as well
keyword duration is 3 and duration is 2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 36492    0   219  100 36273     51   8475  0:00:04  0:00:04 --:--:--  8476
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 28486    0   161  100 28325     43   7691  0:00:03  0:00:03 --:--:--  7690
Attempting to answer: Monty Python
Input interpretation : Monty Python's Flying Circus  (television series)
Series information : title | Monty Python's Flying Circus
network | BBC1  |  BBC2
air dates | 5 October 1969  to  5 December 1974

The above was initiated by using my keyword "Gladys", I got the response "Yes Sir" and then Pi goes in to listening mode for a command. As you can see I asked about Monty Python and Wolfram gave me an answer.

Does this help at all?

mathijsvandenhurk, 2016-09-25:
@Colin1964 This was very helpfull! It worked :) Now I can say the keyword and then a command. Thnx a lot. 

But now I face a new bump in the road. I changed the default language to dutch, because even though I like the English language better, to pronounce everything exactly right will get tiresome I think. So I've already changed the language settings in the config file, tts file and voicecommand.cpp (basically everywhere I saw en of en-gb or en-us) to nl. When I say the keyword in Dutch, it responds with a Dutch voice. Also when I give an improper command, it gives back the response with the Dutch voice. But when I tts something it gives a response in the English voice... Any ideas what might be causing this?

Btw if I don't get this working, I'll just revert to English, but still, gotta try!

Colin1964, 2016-09-25:
@mathijsvandenhurk Excellent! Good to hear. Not sure on the Dutch TTS but if you have used the revised TTS script by @irsx02 in #56, the TTS is hard coded to en-us - regardless of what lang is set elsewhere. Might be worth looking there first?

irsx02, 2016-09-26:
@mathijsvandenhurk  Yes, I got a bit lazy and hard coded TTS to always reply in English.  If you go to the TTS script in #56 (search for the comment "US English, mofo, do you speak it")    Change the line where it says  `lang="en-US"` to ... umm.. well.. I'm not sure what Dutch is, but change it to the setting that is needed for Dutch.

mathijsvandenhurk, 2016-09-28:
@irsx02 @Colin1964 Thnx guys, I've been a bit busy the last couple of days. I will probably look into it this weekend. I'm not using the revised script you mentioned. I can tray that, but I will have to get pico2wave right?

Colin1964, 2016-09-30:
@mathijsvandenhurk Yes - if you use the TTS script by @irsx02 you will need to install pico2wave.

renaldodjunaedi, 2016-12-16:
hii all..
i am beginner on this project but i have succes on install @StevenHickson voice recog.. was able to recognize the voice and control electronic equipment but i have same problem with keyword..

@Colin1964 Can you explain in detail how you use the wolfram to make keyword "Gladys"?? sorry I can not practice just with your explanation above like @mathijsvandenhurk 

hope you guys reply.. thx a b4..

Colin1964, 2016-12-17:
@renaldodjunaedi Wolfram set up is covered in my post in #66 but this won't fix the Keyword not being recognized or change the Keyword to "Gladys". The Keyword is set in the .commands.conf file.  Are you running with` !continuous==1` or `!continuous==0`? What output are you getting, both displayed and audio, when you say your keyword? Is `!verify=1`?