StevenHickson/PiAUISuite #79: Unable to install voicecommand due to missing library

chrisspen, 2016-10-22:
Attempting to install voicecommand on Ubuntu 16 fails with the error:

     voicecommand: error while loading shared libraries: libboost_regex.so.1.54.0: cannot open shared object file: No such file or directory

even though I installed all listed dependencies.

chrisspen, 2016-10-22:
I suspect the issue is because I'm on Ubuntu 16 and I'm using your pre-compiled binary which dynamically links and was likely built on an older distro. Ubuntu 16 has a similar library, but it's called `libboost_regex.so.1.58.0`.

The obvious solution is to recompile, however, your makefile is incomplete and doesn't work on x86. Attempting to `make` gives me the errors and warnings:

    g++ -m32 -D'BUILDTS="161021 22:04:08 -0400"' -O3 -lcurl -lboost_regex -o x86/voicecommand voicecommand.cpp
    In file included from /usr/include/curl/curl.h:35:0,
                     from voicecommand.h:15,
                     from voicecommand.cpp:1:
    /usr/include/curl/curlrules.h:143:41: error: size of array '__curl_rule_01__' is negative
         [CurlchkszEQ(long, CURL_SIZEOF_LONG)];
                                             ^
    /usr/include/curl/curlrules.h:153:53: error: size of array '__curl_rule_02__' is negative
         [CurlchkszEQ(curl_off_t, CURL_SIZEOF_CURL_OFF_T)];
                                                         ^
    voicecommand.cpp: In member function 'void VoiceCommand::ProcessMessage(const char*)':
    voicecommand.cpp:371:44: warning: format not a string literal and no format arguments [-Wformat-security]
             fprintf(stderr,improper_tmp.c_str());
                                                ^
    voicecommand.cpp: In function 'int main(int, char**)':
    voicecommand.cpp:51:45: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
         system("echo \"\" > /dev/shm/voice.log"); //lazily clear out the log file
                                                 ^
    voicecommand.cpp:93:151: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
                         system("flac /dev/shm/noise.wav -f --best --sample-rate 16000 -o /dev/shm/noise.flac 1>>/dev/shm/voice.log 2>>/dev/shm/voice.log");
                                                                                                                                                           ^
    voicecommand.cpp:97:55: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
                         fscanf(cmd,"\"%[^\"\n]\"",message);
                                                           ^
    voicecommand.cpp:99:54: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
                         system("rm -fr /dev/shm/noise.*");
                                                          ^
    voicecommand.cpp:138:47: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
                 fscanf(cmd,"\"%[^\"\n]\"",message);
                                                   ^
    voicecommand.cpp: In function 'float GetVolume(std::__cxx11::string, std::__cxx11::string, bool)':
    voicecommand.cpp:39:24: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
         system(run.c_str());
                            ^
    voicecommand.cpp:41:26: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
         fscanf(cmd,"%f",&vol);
                              ^
    voicecommand.cpp: In function 'void ProcessVoice(FILE*, VoiceCommand&, char*)':
    voicecommand.cpp:24:39: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
         fscanf(cmd,"\"%[^\"\n]\"",message);
                                           ^
    voicecommand.cpp: In member function 'void VoiceCommand::ProcessMessage(const char*)':
    voicecommand.cpp:307:47: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
                         system(newcommand.c_str());
                                                   ^
    voicecommand.cpp:314:40: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
                         system(tmp.c_str());
                                            ^
    voicecommand.cpp:329:34: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
           system(commands[i].c_str());
                                      ^
    voicecommand.cpp:361:44: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
                             system(run.c_str());
                                                ^
    voicecommand.cpp: In member function 'void VoiceCommand::DisplayUsage()':
    voicecommand.cpp:246:31: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
         system("man voicecommand");
                                   ^
    voicecommand.cpp: In member function 'void VoiceCommand::EditConfig()':
    voicecommand.cpp:489:33: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
         system(edit_command.c_str());
[... cut ...]
    voicecommand.cpp:949:35: warning: ignoring return value of 'int scanf(const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
                     scanf("%s",buffer);
                                       ^
    voicecommand.cpp:965:24: warning: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Wunused-result]
         system(tmp.c_str());
                            ^
    Makefile:31: recipe for target 'voicecommand' failed
    make: *** [voicecommand] Error 1

Colin1964, 2016-10-22:
@chrisspen Have you tried the solution described in #58 ? 
This solution here was based on libboost 1.55.1 as the latest version so you may either need to apt-get 1.55.1 (if you can?) and use the command line instruction as is or alter the original command line instruction to read (if you have 1.58.0) "1.58.0" ie.

Create a symlink called libboost_regex.so.1.46.1 that points at libboost_regex.so.1.58.0 with:

`ln -s /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.46.1`

NoBreakz, 2017-03-07:
collin1964: thanx man, i had the same problem, think it fixed it ;)

jamieduk, 2017-07-12:
voicecommand: error while loading shared libraries: libboost_regex.so.1.54.0: cannot open shared object file: No such file or directory

still get error and dont know why :/ im using mint 18.2

HamidSayeed, 2018-07-17:
I have same error , any help appreciated .
voicecommand: error while loading shared libraries: libboost_regex.so.1.54.0: cannot open shared object file: No such file or directory