StevenHickson/PiAUISuite #80: Microphone and audio out problem

FlyManCz, 2016-10-27:
When i run the command voicecommand -s
Then i go through the process. When it asks "First I'm  going to say something and see if you hear it" i dont hear anything. When i try alsamixer command it doesn't do anything :/

If i ignore the audio out problem i get another problem when it asks "Do you want to set up and check the speech recognition options?" then it says "sh: 1: arecord: not found
sh: 1: arecord: not found
I couldn't find a hardware device. You don't have a valid microphone"

What should i do?
I am really confused.

seky2205, 2016-11-09:
I have found out that link used in tts script is no longer valid.
After running: 
wget -U Mozilla -O "/dev/shm/tmp.mp3" "https://translate.google.com/translate_tts?tl=en&q=good&ie=UTF-8&total=1&idx=0&client=t"
I get:
--2016-11-09 14:50:59--  https://translate.google.com/translate_tts?tl=en&q=good&ie=UTF-8&total=1&idx=0&client=t
Resolving translate.google.com (translate.google.com)... 216.58.209.174, 2a00:1450:400d:806::200e
Connecting to translate.google.com (translate.google.com)|216.58.209.174|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2016-11-09 14:51:00 ERROR 403: Forbidden.
And after going to same URL with desktop browser I get 403 missing permissions

seky2205, 2016-11-09:
I found out you need to change it. Instead of client=t it needs to be client=tw-ob 
http://stackoverflow.com/questions/32053442/google-translate-tts-api-blocked

BHEADRICK, 2017-01-09:
that's correct, that does look like it needs to be changed. This works:
https://translate.google.com/translate_tts?tl=EN&q=hello&ie=UTF-8&total=1&idx=0&client=tw-ob

However, google is blocking automated requests to the tts service, however, there is a workaround:
https://community.openhab.org/t/googletts-responding-with-503-error-even-after-the-url-fix/3385/3

BHEADRICK, 2017-01-09:
this is what I changed my tts script to: 
You'll have to add your own api key, of course. 
```
#!/bin/bash

#for the Raspberry Pi, we need to insert some sort of FILLER here since it cuts off the first bit of audio

string=$@
lang="enous"
echo "" > "/dev/shm/speak.mp3"

len=${#string}

#this will get the last remnants
wget -q -U Mozilla -O "/dev/shm/tmp.mp3" "http://api.voicerss.org/?key=<API key>&hl=${lang}&src=$string"
cat "/dev/shm/tmp.mp3" >> "/dev/shm/speak.mp3"
#now we finally say the whole thing
cat "/dev/shm/speak.mp3" | mpg123 - 1>>/dev/shm/voice.log 2>>/dev/shm/voice.log
:q6+
```

GC-Swift2, 2017-01-13:
@BHEADRICK
Did you get the api key from this Webseite http://www.voicerss.org/
And does it work now?
I tried the same but in my case it doesn't work.