StevenHickson/PiAUISuite #88: Verify==1 keyword not working

symatix, 2017-02-12:
Hello Steven! First I want to say - awesome package! 👍 
But I am having trouble making keyword verification to work. So far I have changed tts to pico2wave so I am getting sound if verification is set to 0. The commands also work, but it is triggered by every sound and it is not usable if I don't look at the monitor to see if it is ready for input or it is already processing something. If I put !verify==1 in to conf file, there is no audio out coming from voicecommand and it starts to continuously record. Of course, multiple attempts of saying the keyword, from whisper to shouting yielded no result..

My conf file is
```
!continuous==1
!verify==1
!ignore==1
!filler==0
!response==Master?
!thresh==1.2
!keyword==anybody
~music==mpsyt playurl https://www.youtube.com/watch?v=ljQsRLN2dXA
~stop==sudo pkill mpsyt
~ressurect==tts "I live!"
```

when verify is set to 0, the output is this
```
Found audio
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 62561    0   436  100 62125    187  26761  0:00:02  0:00:02 --:--:-- 26778
command: mpsyt playurl https://www.youtube.com/watch?v=ljQsRLN2dXA
Chillhop Cafe · jazzy ' lofi hip hop radio  · beats for studying / work
```
but when it is set to 1, it will always recognize audio input, but not the keyword, so the output looks like this
```
Opening config file...
Not querying for answers
running in continuous mode
verifying command as well
keyword duration is 2 and duration is 3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:51 --:--:--     0
```

symatix, 2017-02-15:
I have found a workaround. Installed [amixer-webui](https://github.com/JiriSko/amixer-webui) to control the mic from the web ui, set `!verify=0 !continuous==1` so now it will begin to record only when i enable the mic. This is even better solution for privacy because it does not send audio stream to google in an effort to recognize the keyword while I'm talking in the house. Now to go and code a one button solution for that, and I'm good to go :)

But I would still like to know the answer to my original post, as it was a problem that I spent hours before giving up.

awaisahmed119, 2018-02-09:
Im having the same issue. If someone know the solution then plz post it here.

awaisahmed119, 2018-02-09:
@symatix did you find the solution for this problem?