StevenHickson/PiAUISuite #89: No translation , No output

Kartikkh, 2017-03-06:
I am getting a weird error of "No translation" when I use voicecommand -c for giving voice commands.
it only ran for 1-2 sec and gave an error of "NO translation "
pi@raspberrypi:~ $ sudo voicecommand -c
Opening config file…
running in continuous mode
keyword duration is 2 and duration is 3
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 15149 0 14 100 15135 4 4793 0:00:03 0:00:03 –:–:– 4795
No translation
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 15190 0 14 100 15176 3 3609 0:00:04 0:00:04 –:–:– 3912
No translation
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 15294 0 14 100 15280 6 6735 0:00:02 0:00:02 –:–:– 6737
No translation
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 15244 0 14 100 15230 2 2441 0:00:06 0:00:06 –:–:– 4088
No translation
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 16830 0 14 100 16816 5 7153 0:00:02 0:00:02 –:–:– 7152
No translation
Found audio
Recording WAVE ‘stdin’ : Unsigned 8 bit, Rate 16000 Hz, Mono
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 –:–:– –:–:– –:–:– 0
[6]+ Stopped sudo voicecommand -c
pi@raspberrypi:~ $ sudo voicecommand -c

MomchilGorchev, 2017-03-18:
Having the same problem on RaspberryPi 3 running Raspbian
Recording sounds with `arecord -D plughw:1,0 test.wav` and then playing them using `aplay test.wav` works fine

Kartikkh, 2017-03-18:
I think due to change in API by google (which I have read in many blogs and pages ). we are recurring through this problem! 
so it would be better to switch to another thing like Jasper etc