StevenHickson/PiAUISuite #94: [Raspberry]error while loading shared libraries...

Franzi98, 2017-05-27:
Hello, I have this problem:
On my rasp I have installed voicecommand and all the libraries but when I try to launch the script the terminal say me "error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory"
What I have to do?
PS: Sorry for my english but I'm Italian

Franzi98, 2017-05-27:
Ok, now it works... 
sudo apt-get install libboost-regex1.49.0

malacanang, 2017-08-03:
did that really work for you? I feel like I've tried everything, when I originally tried that the library was no longer available.

fransafu, 2017-08-12:
Hello, I had the same problem and looking for the problem come to this solution:

1) search and identify in the **"/usr/lib/x86_64-linux-gnu"** folder the **"libboost"** library you have.

Use the following command:

`ls /usr/lib/x86_64-linux-gnu | grep 'libboost_regex'`

> OUTPUT:
>
> libboost_regex.a
> libboost_regex.so
> libboost_regex.so.1.58.0

2) Look at what libreria is requesting in the error, in my case: "voicecommand: error while loading shared libraries: libboost_regex.so.1.54.0: can not open shared object file: No such file or directory". voicecommand request: "libboost_regex.so.1.54.0".

3) Create link, use the following command:

`sudo ln -s /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.54.0`

4) And this command (refresh Shell):

`$SHELL`

**Remember to replace the library you have at the beginning and in the end put the name of the library that requests the program.**

This is a patch solution, but it helps to get out of trouble.