StevenHickson/PiAUISuite #97: voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory

rottieGO, 2017-10-25:
1.  I'm running Raspberian Stretch on Rasbperry Pi 3b
2.  Followed instruction from https://diyhacking.com/best-voice-recognition-software-for-raspberry-pi/ but failed at the end of step 2 when it gets to VoiceCommand section with error

Would you like voicecommand to try to set itself up? y/n
voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory

3. When executing command:  

sudo apt-get install -y libboost-dev libboost-regex-dev youtube-dl axel curl xterm libcurl4-gnutls-dev mpg123 flac sox

results are:
axel is already the newest version (2.12-1).
curl is already the newest version (7.52.1-5+deb9u1).
flac is already the newest version (1.3.2-1).
libboost-dev is already the newest version (1.62.0.1+b4).
libboost-regex-dev is already the newest version (1.62.0.1+b4).
libcurl4-gnutls-dev is already the newest version (7.52.1-5+deb9u1).
mpg123 is already the newest version (1.23.8-1).
sox is already the newest version (14.4.1-5).
xterm is already the newest version (327-2).
youtube-dl is already the newest version (2017.05.18.1-1).

host 9.10.3-P4-Raspbian

I'm pretty certain its because of Stretch vs Jessie but I'm not Linux savvy enough to figure out what and where to change.

Are you working on updating the code by any chance or did anyone figure out how to solve this?
I read on one post to do the below but gives me no results, in fact my /usr/lib doesn't have x86_64-linux-gnu directory

ls /usr/lib/x86_64-linux-gnu | grep 'libboost_regex'

Any help is greatly appreciated.

AbuTaj, 2017-11-15:
I too have a same issue. Any body found a solution?

moltak, 2017-12-23:
This is related with version of g++, clone this https://github.com/moltak/PiAUISuite or edit `VoiceCommand/Makefile` at 11line `g++-4.8` to `g++`

moltak, 2017-12-23:
or type `sudo apt-get install libboost1.49-dev libboost-regex1.49-dev`

transmit-ter, 2017-12-31:
dear @moltak
i did  clone this https://github.com/moltak/PiAUISuite  and try type sudo apt-get install libboost1.49-dev libboost-regex1.49-dev like you posted but its not woking
still have
voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory
Any suggestion?

transmit-ter, 2018-01-02:
You may want to try this
git clone git://github.com/StevenHickson/PiAUISuite.git
cd PiAUISuite/Install/
./InstallAUISuite.sh
cd PiAUISuite/VoiceCommand
make clean
g++ -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="160506 05:20:31 -0700"' -O3 -lcurl -lboost_regex -o voicecommand voicecommand.cpp
cp voicecommand /usr/bin
voicecommand -s

remohn, 2018-02-15:
I get to the voicecommand setup and then don't hear anything. Does voicecommand work through a bluetooth speaker?

s1h4d0w, 2018-02-23:
@transmit-ter @moltak 

Both fixes don't work for me, I've tried numerous times, including other possible fixes such as alternate repositories for packages and trying to use a different version of libboost_regex.

I've tried transmit-ter's approach and it fails on "make clean":

[image]

I tried skipping it but the next command also gives errors:

[image]

I have the exact same problem as @jojosiao and I've been trying for weeks to get this to work, but I can't make it work. It's missing libboost_regex.so.1.49.0 and I can't for the life of me figure out how to get it or how to replace it. I tried different versions of libboost_regex, doesn't work. I tried getting the original 1.49.0 package elsewhere, no luck. It's driving me crazy.

[image]

boldra, 2018-08-30:
Symlinking the libraries (as per s1h4d0ws raspi forum link above) worked for me, but it was also necessary to **rerun the PiAUI install script after creating the symlink**

KarelKubat, 2019-01-06:
Symlinking different versions of shared libs isn't the solution. You will get a loadable binary, but it won't do anything useful unless those shared libs are really pretty much identical - which they probably won't be (otherwise, why the version difference?). 

Trying out Matthew Larkin's solution doesn't give a working solution, at least, not for me. 

The thing to do is to
- clone https://github.com/StevenHickson/PiAUISuite
- remove the stale binaries in the repo that have been built against older-version libs; meaning: `make clean` where appropriate, in this case in `PiAUISuite/VoiceCommand`
- fix up `PiAUISuite/VoiceCommand/Makefile` to use compiler `g++` instead of `g++-4.8`
- Re-run `make`
- Manually copy `voicecommand` to `/usr/bin/`

However, looking at the source, I get a feeling this C++ version of `voicecommand` can be pretty much replaced by a scripted version that's independent of libs and whatnot - it's probably better off in Python. It's basically just config file parsing and execing a bunch of standard Unix commands. TBH I don't see the need why this should be expressed in C++ which introduces binary dependencies.

s1h4d0w, 2019-01-06:
/\ This exactly.

I've messed with symlinking other versions but  it simply does not work. I haven't gotten VoiceCommand to work even once, and believe me, I've spent hours on it.

thepony, 2019-01-07:
I tried touching base with the developer (no response... *shrugs*) so I did a bunch of leg work for this to find a workable solution. After about a week of testing I came up with a workable solution with a very simple python script. As long as your audio devices are correctly set up you can use it https://www.github.com/thepony/piTalk

S-Huzaifa, 2020-08-18:
Assalam u Alikum
This is not a big problem I also got this.
When you install PiAUISuite (by /InstallAUISuite.sh)
it asks for dependencies I think you don't allow this.
you have to reinstall it.