StevenHickson/PiAUISuite #98: noise.flac

winnerke, 2017-11-03:
Hey!
When I start voicecommand receive this warning in every 2-3 seconds. 

> Warning: Couldn't read data from file "/dev/shm/noise.flac", this makes an  Warning: empty POST.

And also when I try to re-run the configuration I didn't hear any voice,
My ,asoundrc is correct. 

And the Suite is up-to-date.

Any suggestion?

Lewiseab, 2017-11-13:
I'm experiencing the same symptoms you mention and can not yet figure out a solution.

KDMcMullan, 2017-11-22:
Ah. I think this is the same problem as issue #99.

sebastw22, 2018-02-03:
I have the same issue. Anyone solved it?

dawsondeere, 2018-03-19:
After looking at /dev/shm/voice.log and seeing the following error, I found out that there's a bad flag for the flac command
`ERROR: raw format options (--endian, --sign, --channels, --bps, and --sample-rate) are not allowed for non-raw input`
On line 93 of voicecommand.cpp
`system("flac /dev/shm/noise.wav -f --best --sample-rate 16000 -o /dev/shm/noise.flac 1>>/dev/shm/voice.log 2>>/dev/shm/voice.log");`
Get rid of `--sample-rate 1600` and recompile, and it should work
Also, on line 35 of speech-recog.sh, the same bad flag is there which causes the error with out.flac, so you'll need to remove it there too
Should also solve #99 and #100

arunvramesh, 2018-03-23:
@dawsondeere  what do you mean by recompile the voicecommand.cpp.I Am a newbie. please explain.

dawsondeere, 2018-03-23:
@arunvramesh When I say recompile voicecommand.cpp, I mean from the directory PiAUISuite/VoiceCommand, run the command `make voicecommand` and when that is finished, run `cp voicecommand /usr/bin/voicecommand` (this command will probably need to be prefixed with `sudo`)

IamVindicta, 2018-03-29:
@dawsondeere when i run the make command i get this, any help?

make voicecommand
g++-4.8 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="180329 04:50:54 +0100"' -O3 -lcurl -lboost_regex -o voicecommand voicecommand.cpp
make: g++-4.8: Command not found
Makefile:34: recipe for target 'voicecommand' failed
make: *** [voicecommand] Error 127

dawsondeere, 2018-03-29:
@IamVindicta You don't have g++-4.8 installed. Change `g++-4.8` to `g++` and then run it (assuming you have g++ installed)

IamVindicta, 2018-03-29:
@dawsondeere i installed g++-4.8 and ran again but this time i ended with a different error, sorry im not the best with this.

make voicecommand
g++-4.8 mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="180329 18:11:11 +0100"' -O3 -lcurl -lboost_regex -o voicecommand voicecommand.cpp
g++-4.8: error: mcpu=cortex-a7: No such file or directory
Makefile:34: recipe for target 'voicecommand' failed
make: *** [voicecommand] Error 1

IamVindicta, 2018-03-29:
@dawsondeere would it be possible for you to upload the complied files somewhere for me and others having issues to download and replace the originals?

Hazhan, 2018-10-24:
@IamVindicta Thank you a lot, I could reach this point with your help but still I have this problem

g++-4.8 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="181024 22:39:29 +0100"' -O3 -lcurl -lboost_regex -o voicecommand voicecommand.cpp
^CMakefile:34: recipe for target 'voicecommand' failed
make: *** [voicecommand] Interrupt

jtsSTECH, 2019-04-16:
> @IamVindicta Thank you a lot, I could reach this point with your help but still I have this problem
>
> g++-4.8 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="181024 22:39:29 +0100"' -O3 [...]

The make file is using g++4.8.  Try compiling it with your main g++ version:
g++ -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -D'BUILDTS="160506 05:20:31 -0700"' -O3 -lcurl -lboost_regex -o voicecommand 

(Found here: https://www.raspberrypi.org/forums/viewtopic.php?t=72570)

borhanshuvo, 2019-12-22:
Warning: Couldn't read data from file "/dev/shm/noise.flac", this makes an
Warning: empty POST. how to solve