StevenHickson/PiAUISuite #99: /dev/shm/out.flac not found

KDMcMullan, 2017-11-22:
Attempts to execute
`voicecommand -c
`
yield up the below:

```
Opening config file...
running in quiet mode
Not querying for answers
running in continuous mode
keyword duration is 2 and duration is 3
Found audio
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 16000 Hz, Mono
Warning: Couldn't read data from file "/dev/shm/out.flac", this makes an empty
Warning: POST.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    14    0    14    0     0     36      0 --:--:-- --:--:-- --:--:--    36
rm: cannot remove '/dev/shm/out.flac': No such file or directory
voicecommand: symbol lookup error: voicecommand: undefined symbol: _ZN5boost9re_detail12perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE14construct_initERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsE

```
I wondered if _ZN5boost9re might be something to do with a missing package. Sure enough,
`sudo apt-get install -y libcurl4-gnutls-dev
`
yields up the error:
```
Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libcurl3-gnutls armhf 7.52.1-5+deb9u1
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libcurl4-gnutls-dev armhf 7.52.1-5+deb9u1
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/curl/libcurl3-gnutls_7.52.1-5+deb9u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/curl/libcurl4-gnutls-dev_7.52.1-5+deb9u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

```
Am I barking up the wrong tree? Why does /dev/shm/out.flac seem not to be created?

Thanks.

Basavaraja-MS, 2017-12-21:
I'm also facing the same issue, any workaround found?

s1h4d0w, 2018-02-23:
Responding to say I have the same issue, after finally fixing the libboost-regex.1.49.0 error.

sdusik, 2018-02-24:
The same issue:
voicecommand: symbol lookup error: voicecommand: undefined symbol: _ZN5boost9re_detail12perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE14construct_initERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsE

CBMPi, 2018-05-01:
The same issue: news from anybody?

s1h4d0w, 2018-12-11:
This project seems to be dead. No commits since 2016.

charlesbauer, 2019-04-12:
Open speech-recog.sg and on line 35 remove the option --sample-rate 16000 from the option for the command flac.

alokme123, 2020-04-12:
Thank you!! That helped!