StevenHickson/PiAUISuite #101: error while loading shared libraries: libboost_regex.so.1.54.0: cannot open shared object file: No such file or directory in debian

jojosiao, 2018-01-04:
I tried to install in debian 9.3 but I encountered the error message when I was asked if I wanted to install voicecommand.

I found a command file
Would you like voicecommand to try to set itself up? y/n
y
voicecommand: error while loading shared libraries: libboost_regex.so.1.54.0: cannot open shared object file: No such file or directory

transmit-ter, 2018-01-05:
#97

transmit-ter, 2018-01-05:
you may want to see issue #97

s1h4d0w, 2018-02-23:
@transmit-ter No, because none of the answers there work. I've tried your approach and it fails on "make clean".

[image]

I tried skipping it but the next command also gives errors:

[image]

I have the exact same problem as @jojosiao and I've been trying for weeks to get this to work, but I can't make it work. It's missing libboost_regex.so.1.49.0 and I can't for the life of me figure out how to get it or how to replace it. I tried different versions of libboost_regex, doesn't work. I tried getting the original 1.49.0 package elsewhere, no luck. It's driving me crazy.

[image]

Calliygous87, 2019-07-23:
> voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or directory
pi@raspberrypi:~/Desktop/PiAUISuite/Install $ sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.62.0 /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.49.0
ln: failed to create symbolic link '/usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.49.0': File exists

This is what I get after following the instructions in that link, and then attempting to install again.