StevenHickson/PiAUISuite #109: Install script failing with 'No such file or directory' errors

nospoon, 2019-11-16:
When trying to install the suite, I get the following output:

```
Install voicecommand? y/n
This is probably the coolest script here and ties many of these together. It is an easily customizable voice control system. It uses speech recognition and text to speech to listen to you, respond to you, and run commands based on what you say.
y
cp: cannot stat '../VoiceCommand/voicecommand': No such file or directory
cp: cannot stat '../VoiceCommand/google': No such file or directory
cp: cannot stat '../VoiceCommand/tts': No such file or directory
cp: cannot stat '../VoiceCommand/speech-recog.sh': No such file or directory
cp: cannot stat '../VoiceCommand/voicecommand.8.gz': No such file or directory
No commands found, using default
cp: cannot stat '../VoiceCommand/commands.conf': No such file or directory
Would you like voicecommand to try to set itself up? y/n
y
./Install/InstallAUISuite.sh: line 210: voicecommand: command not found
```

hal8kb, 2020-01-15:
You're executing from wrong directory. Delete your install then type these commands
in your Pi:

git clone https://github.com/StevenHickson/PiAUISuite.git
cd PiAUISuite/Install
./InstallAUISuite.sh

You then need to cd into PiAUISuite and run commands from these

nirvine, 2021-06-07:
I have the same issue with install, but the above answer does not solve it.  The error occurs as a part of the install process. 

From my install log: 

> Install voicecommand? y/n
> This is probably the coolest script here and ties many of these together. It is an easily customizable voice control system. It uses speech recognition and text to speech to listen to you, respond to you, and run commands based on what you say.
> y
> cp: cannot stat '../VoiceCommand/voicecommand': No such file or directory
> I found a command file
> Would you like voicecommand to try to set itself up? y/n
> y
> ./InstallAUISuite.sh: line 210: voicecommand: command not found
If I then go to the PiAUISuite directory and run the command from there, I get:

> pi@raspberrypi:~/PiAUISuite $ ./voicecommand -s
> -bash: ./voicecommand: No such file or directory

There does not appear to be any file in this directory or sub-directories named "voicecommand"

How do I get this to work?