StevenHickson/PiAUISuite #110: Voicecommand: no such file or directory

JakobDeHertogh, 2020-04-13:
[image]

"Voicecommand: no such file or directory", even though the executable is (seemingly) correctly installed. 
Apparently the main reasons this could occur is when it's a 32-bit executable or when there is no interpreter. Neither of those is the case here.

JakobDeHertogh, 2020-04-13:
Okay I found that I do not have the /lib64 directory, and therefore not the correct interpreter. Will update ..

S-Huzaifa, 2020-08-18:
Assalam u Alikum
This is not a big problem I also got this.
When you install PiAUISuite (by */InstallAUISuite.sh*)
it asks for dependencies I think you don't allow this.
you have to reinstall it.