StevenHickson/PiAUISuite #113: Error while installing voicecommand.

Aht-e-sham, 2022-01-07:
I can't install voicecommand. 
voicecommand: error while loading shared libraries: libboost_regex.so.1.49.0: cannot open shared object file: No such file or director
[image]
y

jamiebrowncow, 2022-04-08:
This can be resolved with a symlink to the newer library..
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.49.0

However, then you will hit another issue.. 
voicecommand: /lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by voicecommand)

nexuspcs, 2022-07-09:
> This can be resolved with a symlink to the newer library.. sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.49.0
>
> However, then you [...]

any fix for the voicecommand Curl issue?